lardemua/atom #638: Re-structuring the zau_bot packages

Kazadhum, 2023-10-10:
Hello @miguelriemoliveira and @manuelgitgomes!

As we discussed, I've started to re-structure `zau_bot` so the dependencies are less confusing and the directory structure is more in line with what we usually do. [Here's the repo where I'm doing this](https://github.com/Kazadhum/zau2).

Currently I'm running into the following error, when running `roslaunch zau2_bringup bringup.launch`:

```
[ERROR] [1696930841.597577019]: No name given for the robot.
[robot_state_publisher-4] process has died [pid 7474, exit code 1, cmd /opt/ros/noetic/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/diogo/.ros/log/13344cbe-6751-11ee-bb84-03694608a8eb/robot_state_publisher-4.log].
log file: /home/diogo/.ros/log/13344cbe-6751-11ee-bb84-03694608a8eb/robot_state_publisher-4*.log
```

I won't ask you to go looking through the URDF files extensively yet, but I was wondering if you had any insight as to what might be causing this. I don't think I know where the "name" for the robot is given, apart from the argument in the launch file for `spawn_urdf` node.

Here's the [the launch file](https://github.com/Kazadhum/zau2/blob/main/zau2_bringup/launch/bringup.launch).

manuelgitgomes, 2023-10-10:
Hello @Kazadhum!

It is supposed to be given here:

https://github.com/Kazadhum/zau2/blob/3ecd307e7c4aaf8ed4a4c322631e124f277b12a9/zau2_description/urdf/ur5e.urdf.xacro#L2

An example:

https://github.com/miguelriemoliveira/softbot2/blob/f71dd178b528ceb082d9f6b676665450ecf94269/softbot2_description/urdf/softbot2.urdf.xacro#L2

Kazadhum, 2023-10-10:
Thank you @manuelgitgomes! That worked! I'll keep you updated on the progress.

Kazadhum, 2023-10-11:
Hi @manuelgitgomes and @miguelriemoliveira! I've finally made some progress and it seems like the robotic system is (re)built!

Tomorrow I'll check if the system really is "independent" of others beside the dependencies specified in the README by testing it in a fresh docker container (just to see if the URDFs are as "clean" as possible).

[image]

[image]

For reference, here's the resulting robot's directory tree (2 levels deep):

```
.
├── README.md
├── zau2_bringup
│   ├── CMakeLists.txt
│   ├── include
│   ├── launch
│   ├── package.xml
│   └── src
└── zau2_description
    ├── CMakeLists.txt
    ├── meshes
    ├── models
    ├── package.xml
    ├── rviz
    ├── urdf
    └── worlds

```

I also still need to implement MoveIt, as was done in the original version.

miguelriemoliveira, 2023-10-11:
@Kazadhum , it looks great. I love the display of the camera on the base of the fustrum. Is that gazebo? I did not know it ...

We are moving along ... great!

Kazadhum, 2023-10-11:
@miguelriemoliveira Yes, the first picture is in gazebo! I'm not sure where that comes from, though.

manuelgitgomes, 2023-10-12:
Hello @Kazadhum!

Here is the move_it implementation for universal robots:

https://github.com/ros-industrial/universal_robot

miguelriemoliveira, 2023-10-16:
> Hello @Kazadhum!
> 
> How is the situation here? :-)

Hi @Kazadhum ,

we should start writing the paper next Monday, so  it would be great to have all results collected until then.

From your side is it realistic?

Kazadhum, 2023-10-16:
Hi @manuelgitgomes and @miguelriemoliveira. 

Yes I think it's doable. I'll focus exclusively on this until it is done.

Sorry for the delays.

miguelriemoliveira, 2023-10-16:
> Hi @manuelgitgomes and @miguelriemoliveira.
> 
> Yes I think it's doable. I'll focus exclusively on this until it is done.
> 
> Sorry for the delays.

Thanks. There is no delay, just keep working on it, you're doing great!

Kazadhum, 2023-10-17:
Hello @miguelriemoliveira @manuelgitgomes! Now that the robotic system is up and running, I asked Jorge for the bagfile he used in simulation so I could configure the calibration (e5).
 
Once I got the file, I tried running the configuration, changing the `config.yml` file accordingly. However, I was met with the following error:

```
Configuring calibration package zau2_e5_calibration
_______________________________________________________

Loading config_file /home/diogo/catkin_ws/src/robots/zau2/zau2_e5_calibration/calibration/config.yml
Loading bagfile /home/diogo/bagfiles/zau2/e4_dualrgb_v2.bag
Parsing description file /home/diogo/catkin_ws/src/robots/zau2/zau2_description/urdf/zau2.urdf.xacro
Executing command: xacro /home/diogo/catkin_ws/src/robots/zau2/zau2_description/urdf/zau2.urdf.xacro -o /tmp/description.urdf
Creating transformation tree using the urdf robot description ...
Running several verifications ... please wait ...
Checking if anchored sensor  is valid ... [OK]
Checking for compressed image topics ... 
Topic /camera/rgb/image_raw is in compressed format (/camera/rgb/image_raw/compressed). Will setup a decompressor.
Topic /camera_mb/rgb/image_raw is in compressed format (/camera_mb/rgb/image_raw/compressed). Will setup a decompressor.
Checking for throttled topics ... 
Sensor camera_ee throttled to 10 Hz.
Sensor camera_mb throttled to 10 Hz.
Checking for camera_info topic in bagfile ... [OK]
Checking if coordinate frames from config.yml exist in bagfile ... 
Unknown tag "dynamics" in /robot[@name='zau2']/link[@name='gripper_link']
Unknown tag "hardwareInterface" in /robot[@name='zau2']/transmission[@name='left_finger_trans']/actuator[@name='left_finger_motor']
Unknown tag "hardwareInterface" in /robot[@name='zau2']/transmission[@name='right_finger_trans']/actuator[@name='right_finger_motor']
Traceback (most recent call last):
  File "/home/diogo/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/scripts/configure_calibration_pkg", line 304, in <module>
    paths = nx.shortest_path(gx, config['world_link'], additional_tf['child_link'])
  File "/usr/local/lib/python3.8/dist-packages/networkx/classes/backends.py", line 145, in wrapper
    return func(*args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/networkx/algorithms/shortest_paths/generic.py", line 167, in shortest_path
    paths = nx.bidirectional_shortest_path(G, source, target)
  File "/usr/local/lib/python3.8/dist-packages/networkx/algorithms/shortest_paths/unweighted.py", line 221, in bidirectional_shortest_path
    raise nx.NodeNotFound(msg)
networkx.exception.NodeNotFound: Either source odom or target base_link is not in G

exit status: 1

```

I assume this error means that `odom` isn't found in the transformation tree, but it is visible in RQT when playing the bag file:

[image]

Do you have any idea as to what it might be?

miguelriemoliveira, 2023-10-17:
Hi @Kazadhum ,

Are you running configure with the use_tfs option?

Kazadhum, 2023-10-17:
I wasn't, @miguelriemoliveira! I'll try it out after dinner!

Kazadhum, 2023-10-17:
Config worked when running with `--use_tfs` @miguelriemoliveira! Thank you! I think this issue can be closed now and we'll continue in #627. Do you agree, @miguelriemoliveira and @manuelgitgomes ?

miguelriemoliveira, 2023-10-17:
Please leave it open and assign to me. The error was not explicit...

miguelriemoliveira, 2023-10-18:
Thanks @Kazadhum , can you give me the steps to replicate?

manuelgitgomes, 2023-10-18:
Hello @miguelriemoliveira and @Kazadhum!

I believe the error was given because odom was not on the URDF, so the configuration was unable to find it.
However, odom is on the bag file, so the `-utf` command worked.

Kazadhum, 2023-10-18:
I think that's probably it too @manuelgitgomes.

In order to replicate the error you just need to run `rosrun zau2_e5_calibration configure` (without `-utf`) with the bag file that Jorge gave me. I'll send it to your e-mail via WeTransfer.

miguelriemoliveira, 2023-10-18:
>  I'll send it to your e-mail via WeTransfer.

What will you send?

About giving commands, better in this style:

    rosrun zau2_e5_calibration configure

Don't I have to clone the repository first?

Kazadhum, 2023-10-18:
@miguelriemoliveira Yes sorry, you need to clone my repository:

`git clone https://github.com/Kazadhum/zau2.git`

and then run:

`rosrun zau2_e5_calibration configure`

> What will you send?

I meant the bag file from Jorge. The link to download it should be in your inboxes.

miguelriemoliveira, 2023-10-18:
Hi @Kazadhum ,

I am sorry, can you please send again the bagfile? I lost the email ...

Kazadhum, 2023-10-18:
Yes @miguelriemoliveira, I just re-sent it!

miguelriemoliveira, 2023-10-18:
Thanks. Got it.

miguelriemoliveira, 2023-10-18:
Hi @Kazadhum ,

I am doing this on #652 . Should we close this one?

Kazadhum, 2023-10-18:
Hi @miguelriemoliveira,
Sounds good!

closed 2023-10-18 by Kazadhum