lardemua/atom #640: RGB labelling crashing after PR merging

danifpdra, 2023-10-10:
Hi @manuelgitgomes and @miguelriemoliveira ...

Tired to run the collector after PR merge and it crashed in the detection of RGB. Every other sensor labeling works normally.

Can someone please test this to see if it's just my problem?

Running in larcc with bag https://we.tl/t-EukIpCGU4D

Please pull larcc, the calibration is already configured. RGB labelling crashes when running

`roslaunch larcc_calibration collect_data.launch output_folder:=$ATOM_DATASETS/larcc_subset/train_real/ overwrite:=true`

```
[ERROR] [1696945615.958299, 1663325519.820768]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7fb015e05d30>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 262, in sensorDataReceivedCallback
    self.labelData()  # label the data
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 389, in labelData
    result = self.pattern.detect(image, equalize_histogram=False)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/patterns.py", line 124, in detect
    ccorners = cv2.cornerSubPix(gray, ccorners, (5, 5), (-1, -1), criteria)
cv2.error: OpenCV(4.7.0) /io/opencv/modules/imgproc/src/cornersubpix.cpp:58: error: (-215:Assertion failed) count >= 0 in function 'cornerSubPix'

[ERROR] [1696945622.414972, 1663325526.274324]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7fb0145b1ac0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 262, in sensorDataReceivedCallback
    self.labelData()  # label the data
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 389, in labelData
    result = self.pattern.detect(image, equalize_histogram=False)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/patterns.py", line 124, in detect
    ccorners = cv2.cornerSubPix(gray, ccorners, (5, 5), (-1, -1), criteria)
cv2.error: OpenCV(4.7.0) /io/opencv/modules/imgproc/src/cornersubpix.cpp:58: error: (-215:Assertion failed) count >= 0 in function 'cornerSubPix'

```

danifpdra, 2023-10-10:
If that bag is too small (it might be), here is a bigger one. https://we.tl/t-6VfwW3T0g9

miguelriemoliveira, 2023-10-10:
Hi @danifpdra ,

I confirm the error. Let me check it and get back to you ...

miguelriemoliveira, 2023-10-10:
Hi @danifpdra ,

you have many config.yaml files in larcc. Which one are you using?

miguelriemoliveira, 2023-10-10:
The topic **/camera_4/rgb/camera_info** contains incorrect information.

```terminal
header: 
  seq: 1546
  stamp: 
    secs: 1663325508
    nsecs: 669871863
  frame_id: "camera_4_rgb_optical_frame"
height: 800
width: 1280
distortion_model: ''
D: []
K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

Not sure if this is the problem yet, but this will be a problem later on during calibration.
It must be corrected.

miguelriemoliveira, 2023-10-11:
I will only have to look at this in the afternoon. I will let you know.

danifpdra, 2023-10-11:
> The topic **/camera_4/rgb/camera_info** contains incorrect information.
> 
> ```
> header: 
>   seq: 1546
>   stamp: 
>     secs: 1663325508
>     nsecs: 669871863
>   frame_id: "camera_4_rgb_optical_frame"
> height: 800
> width: 1280
> distortion_model: ''
> D: []
> K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> binning_x: 0
> binning_y: 0
> roi: 
>   x_offset: 0
>   y_offset: 0
>   height: 0
>   width: 0
>   do_rectify: False
> ```
> 
> Not sure if this is the problem yet, but this will be a problem later on during calibration. It must be corrected.

Hi @miguelriemoliveira ,

Here is a small bag with correct intrinsics: https://we.tl/t-kH2gddKZXc

danifpdra, 2023-10-11:
The problem subsists even with fixed rosbag

danifpdra, 2023-10-11:
The problem is here:

https://github.com/lardemua/atom/blob/3c2da29d404d6cbfb55d92ca0369ece3d3c1d09a/atom_calibration/src/atom_calibration/collect/patterns.py#L120-L130

corners exists but ccorners returns None after 

`ret, ccorners, cids = cv2.aruco.interpolateCornersCharuco(corners, ids, gray, self.board)`

which causes the cornerSubPix to fail

danifpdra, 2023-10-11:
This solves the problem 

```
        if len(corners) > 4:
            ret, ccorners, cids = cv2.aruco.interpolateCornersCharuco(corners, ids, gray, self.board)
            criteria = (cv2.TERM_CRITERIA_EPS + cv2.TERM_CRITERIA_MAX_ITER, 500, 0.0001)
            # TODO is it 5x5 or 3x3 ...
            **if ccorners is not None:**
                ccorners = cv2.cornerSubPix(gray, ccorners, (5, 5), (-1, -1), criteria)

            # A valid detection must have at least 25% of the total number of corners.
            detected = ccorners is not None and len(ccorners) > self.number_of_corners / 4

            if detected:
                return {'detected': detected, 'keypoints': ccorners, 'ids': cids.ravel().tolist()}
```

miguelriemoliveira, 2023-10-11:
Hi @danifpdra ,

sorry for the delay, only now I am able to look at this. Your solution works fine? Because if the interpolation is returning none it may be failing. I will make some tests and get back to you.

miguelriemoliveira, 2023-10-11:
Hi @danifpdra ,

thanks for the help. It was the error you found indeed. Since I was looking at the code, I simplified it a little bit and added your suggestion. 

Please stash your changes and pull my changes. It should work fine now ...

closed 2023-10-11 by miguelriemoliveira