lardemua/atom #641: Several crashes in full evaluation

miguelriemoliveira, 2023-10-11:
Hi @manuelgitgomes ,

I was trying to run the full_evaluation for the rgb_rgb_system 

https://github.com/lardemua/atom/tree/noetic-devel/atom_examples/rgb_rgb_system#evaluation

and got several errors

1. If no train or test files are given, the script crashes with an exception. It should say, arguments are required.

2. Error in saveFileResults
```
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/ground_truth_frame_evaluation", line 344, in <module>
    table.printTable()
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/ground_truth_frame_evaluation", line 126, in printTable
    self.table_to_csv()
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/ground_truth_frame_evaluation", line 98, in table_to_csv
    saveFileResults(args['train_json_file'], args['test_json_file'], results_name, self.table_to_save.get_csv_string.get_csv_string()())
```

3.Error in ground truth frame evaluation
```
[ground_truth_frame_evaluation-6] process has died [pid 56896, exit code 1, cmd /home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/ground_truth_frame_evaluation -train_json /home/mike/datasets/rgb_rgb_system/rgb_rgb_system_example_train_dataset/atom_calibration.json -test_json /home/mike/datasets/rgb_rgb_system/rgb_rgb_system_example_test_dataset/dataset.json -csf lambda x: True -sfr __name:=ground_truth_frame_evaluation __log:=/home/mike/.ros/log/24c69e42-6890-11ee-afde-d3b9432cd01b/ground_truth_frame_evaluation-6.log].
log file: /home/mike/.ros/log/24c69e42-6890-11ee-afde-d3b9432cd01b/ground_truth_frame_evaluation-6*.log
```

manuelgitgomes, 2023-10-13:
Hello @miguelriemoliveira!

Responding to your points:

> 1. If no train or test files are given, the script crashes with an exception. It should say, arguments are required.

I believe required arguments are not a function inside launch files, so I cannot do it from there.
On the nodes, it is already implemented.
However, as all nodes are running simultaneously, the messages get all jumbled up and it might not be trivial to understand.
As an examples, various messages from the nodes saying `error: argument -train_json/--train_json_file: expected one argument`:
[image]

> 2\. Error in saveFileResults
>
>
> ```
> Traceback (most recent call last):
>   File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/ground_truth_frame_evaluation", line 344, [...]

Error 2 and 3 are dependent (2 is the error on the script and 3 is the node running that script crashing).

I have changed it and it is now working.
However, we have some funny intercollection errors, I will investigate it:

[image]

manuelgitgomes, 2023-10-13:
Hello @miguelriemoliveira!

This seems to me like a conceptual mistake we have made.

When we created inter collection, I believe we might have made a lapse in judgement, as it only works when the pattern is fixed.
This is because we project the detection from the sensor source to the world frame in the source collection and then project it to sensor target from the world frame in the target collection and then compare them.
This projection will invariably end up at the same global position in the target collection as in the source collection.

Therefore, if the pattern move between collections, this breaks, right?

miguelriemoliveira, 2023-10-13:
Hi @manuelgitgomes ,

yes, I agree. The pattern must be fixed to compute inter-collection errors.

You can check the json configuration, it has somewhere a field that says if the pattern is fixed.

We can compute only if that is true ...

manuelgitgomes, 2023-10-16:
Hello @miguelriemoliveira!

I have solved this by creating the following function:

https://github.com/lardemua/atom/blob/1b6fe25ff717a4dd771e7857be7d49294322df2b/atom_core/src/atom_core/utilities.py#L191-L195

This function is ran before every inter collection script and we are unable to compute them.

I have also added a condition to the calibration configuration so that, when the pattern is not fixed, inter collection evaluation is not added to `full_evaluation.launch`.
The full evaluation of the `rgb_rgb_system` now carries out flawlessly!

miguelriemoliveira, 2023-10-16:
Thanks I will test from my side

miguelriemoliveira, 2023-10-18:
Hi @manuelgitgomes ,

I tested and now it does not crash. However, a strange behavior occurs. The launch file never completes. Look at the cursor at the bottom.

[image]

Once the evaluations are finished, the script should exit.

Another thing is that its not easy to understand what each table means. Perhaps we should add a print that explains exactly what each table is. What do you think?

manuelgitgomes, 2023-10-18:
Hello @miguelriemoliveira!

Ok, I will investigate how that is done!

I think the explanation is only doable if we add it to the evaluation itself. A title for the table, perhaps.

miguelriemoliveira, 2023-10-18:
> I think the explanation is only doable if we add it to the evaluation itself. A title for the table, perhaps.

I agree. Give me an example and we can brainstorm about the best text.

miguelriemoliveira, 2023-10-22:
... and while we do this might as well solve #622 on the same go.

manuelgitgomes, 2023-11-22:
> Once the evaluations are finished, the script should exit.

I don't know what has changed between this and now, but I cannot seem to replicate this bug. 
Can you test on your side and verify?

miguelriemoliveira, 2023-11-29:
Will test later on and create another issue if problem persists. Thanks

Closing.

closed 2023-11-29 by miguelriemoliveira