lardemua/atom #642: Calibrate rgb_depth_system

miguelriemoliveira, 2023-10-12:
Tagging @brunofavs and @danifpdra for visibility.

The goal is to calibrate rg_depth_system and write a tutorial on how to do it.

miguelriemoliveira, 2023-10-12:
Have some problems with the labeling of depth data, as was reported by @danifpdra in #612.

Will solve that one first and get back to this.

miguelriemoliveira, 2023-10-22:
#612 is solved, so coming back to this one.

miguelriemoliveira, 2023-10-22:
@brunofavs , we should meet soon. When are you available?

brunofavs, 2023-10-22:
Hey
I havent yet had time to reinstall my OS to sort out the ros dependencies problem, mainly because of other projects.

Can we postpone this meeting one week until the 2 projects I have to finish for other courses are over?

miguelriemoliveira, 2023-10-22:
Sure. Let me know when you have time.

miguelriemoliveira, 2023-10-28:
Created a calibration tutorial  and calibrated. 

I think all is done now. 

@danifpdra , can you please take a look at the readme, and give a critical opinion of the readme.

In particular, the final evaluation average of 2 pixels ... do you think its ok?
https://github.com/lardemua/atom/tree/noetic-devel/atom_examples/rgb_depth_system

miguelriemoliveira, 2023-10-29:
This is done for now, except for a few questions about the final evaluation results (#672).

In any case, @brunofavs , you can try it out.

closed 2023-10-29 by miguelriemoliveira