lardemua/atom #644: Automatic depth labeling not excluding image edges

danifpdra, 2023-10-13:
[image]

danifpdra, 2023-10-13:
We should add some "rubber" to delete faulty edges

danifpdra, 2023-10-13:
The problem seems to be inside the function labelDepthMessage

danifpdra, 2023-10-13:
Changed the filter_border_edges in the function arguments and seems to solve the problem?

```
        labels, gui_image, _ = labelDepthMsg(msg, seed=None, bridge=None,
                                             pyrdown=0, scatter_seed=True,
                                             scatter_seed_radius=8,
                                             debug=False,
                                             subsample_solid_points=7, limit_sample_step=1,
                                             filter_border_edges=0.05,
                                             pattern_mask=pattern_mask)

```

This argument wasn't set for inside the depth_manual_labeling.py script

danifpdra, 2023-10-13:
Solves the problem for vertical limits but not for horizontal limits

[image]

danifpdra, 2023-10-13:
The problem seems to be that the image has some weird black huge edges around the actual image??

[image]

danifpdra, 2023-10-13:
Solved for vertical edges by filtering 7.5% of the image in each border

danifpdra, 2023-10-13:
> Thanks @danifpdra ,
> 
> So the edge near to image boundaries already exists, but the problem now are there gigantic holes in the pattern near the bottom?

No, I think the problem is that the real depth maps have a huge back frame around the actual image

danifpdra, 2023-10-13:
If you see in the image above, I am clicking on the black part in the bottom and side and it still is recognized by the click on image plugin

miguelriemoliveira, 2023-10-13:
Hum, ok. I will investigate.

miguelriemoliveira, 2023-10-14:
Hi @danifpdra ,

I noticed the argument **filter_border_edges=0.05** has a value. 

did you try to change it to see if it works?

danifpdra, 2023-10-14:
> Solved for vertical edges by filtering 7.5% of the image in each border

Yes, I told you here. But the same value doesn't work for all the edges

danifpdra, 2023-10-17:
Hi @miguelriemoliveira,

I told you I would come back with more bugs!

This didn't work for vertical edges

[image]

miguelriemoliveira, 2023-10-17:
There was a bug. I restructured and now I think its working. Can you try it please?

miguelriemoliveira, 2023-10-17:
There a problem though. For simulated images the entire top part of the image is nans. 

I tested with atom_examples/rgb_depth_system

    roslaunch rgb_depth_system_calibration dataset_playback.launch

and then

    rosrun atom_calibration dataset_playback -json $ATOM_DATASETS/rgb_depth_system/rgb_depth_system_example_train_dataset/dataset.json -ow

This is what happens. With this user defined polygon:

[image]

we get this:

[image]

Notice how the upper corner has the border points removed.

 I think there is nothing we can do here. My suggestion is to add a flag in the **labelDepthMsg** called **remove_nan_border** 

If **remove_nan_border** == False  -> run the border detection assuming the entire image canvas, like we did before

If **remove_nan_border** == True  -> run the border detection using the detection of the nan frames around the image

You agree? I added this parameter already to test, but now we must add the functionality to set it from the command line when we call 

    rosrun atom_calibration dataset_playback -json $ATOM_DATASETS/rgb_depth_system/rgb_depth_system_example_train_dataset/dataset.json -ow

If you agree I will do it ...

danifpdra, 2023-10-17:
I think it's a good idea

miguelriemoliveira, 2023-10-17:
Implemented. The argument is called remove_nan_border, you can use it like this:

    rosrun atom_calibration dataset_playback -json $ATOM_DATASETS/rgb_depth_system/rgb_depth_system_example_train_dataset/dataset.json -ow --remove_nan_border

Notice that by default it does not remove the border, i.e., will work poorly for your real images from hell.

closed 2023-10-17 by miguelriemoliveira

danifpdra, 2023-10-17:
It works poorly anyway in the images from hell 🥲

[image]

danifpdra, 2023-10-17:
Something weird happened here...
[image]

commit 2b02fcfc (2023-10-17): Fix #644. using a computation of the used image area.
commit b807c70c (2023-10-17): Close #644