lardemua/atom #650: Import ros_numpy functionalities into ATOM

miguelriemoliveira, 2023-10-16:
In #647 we saw that we have a problem with the version that  ros_numpy inposes for numpy.

We want to get rid of this issue so the suggestion is to copy the code we need from ros_numpy into atom and update to make it work with numpy>1.20.

manuelgitgomes, 2023-10-17:
Hello @miguelriemoliveira.

Ros_numpy is written in a way very different than the one I am used to, and it seemed that it had a large interdependecy between files.

Therefore, I copied all of ros_numpy tand placed it inside atom_core:

https://github.com/lardemua/atom/tree/noetic-devel/atom_core/src/atom_core/ros_numpy

See if this is acceptable in terms of organization.

In terms of functionality, I carried out the calibration and full evaluation of softbot2 and I have no errors to report.

miguelriemoliveira, 2023-10-18:
Not my preferred way but for now I think we can keep it. How about the licenses? Do you see any problem?

manuelgitgomes, 2023-10-19:
Good point, did not think about that.

The license is MIT, one of the most permissive ones.
I have added their license and copyright notice in a LICENSE file inside the ros_numpy directory.

closed 2023-11-14 by miguelriemoliveira