lardemua/atom #651: Error when calibrating larcc subset

danifpdra, 2023-10-18:
Got the following error when running 

`rosrun atom_calibration calibrate -json /home/daniela/datasets/larcc_subset/train_real/dataset_corrected.json -v -uic`

```
Traceback (most recent call last):
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 497, in <module>
    main()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 441, in main
    residuals = objectiveFunction(opt.data_models)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/calibration/objective_function.py", line 692, in objectiveFunction
    m_pt = np.reshape(detected_limit_points_in_pattern[0:2, idx], (1, 2))
IndexError: index 347 is out of bounds for axis 1 with size 345
```

I will investigate...

danifpdra, 2023-10-18:
I think I know the problem, can you zoom @miguelriemoliveira ?

danifpdra, 2023-10-18:
The manual labeler was not deleting indexes with NaN values... Hence this problem https://github.com/lardemua/atom/issues/644#issuecomment-1766763407 and the error I was getting... The calibrate script uses the length of the index points to create the samples and it wasn't able to understand that some of those indexes weren't supposed to be used because their value in the image was NaN... It uses just the length of that list blindly. Anyway this problem was in the labeler part after all and it's now solved

miguelriemoliveira, 2023-10-18:
Hi @danifpdra ,

sorry, only now I am going through the emails. Great that its solved. Can we close it?

miguelriemoliveira, 2023-10-22:
@danifpdra , you never remember to close the issues ... do you have a problem with closing stuff? Are you from Braga :-) ?

closed 2023-10-22 by miguelriemoliveira