lardemua/atom #652: Print informative error when frames are missing from the xacro or the bagfile

miguelriemoliveira, 2023-10-18:
In #638 we got this error:

```
Configuring calibration package zau2_e5_calibration
_______________________________________________________

Loading config_file /home/diogo/catkin_ws/src/robots/zau2/zau2_e5_calibration/calibration/config.yml
Loading bagfile /home/diogo/bagfiles/zau2/e4_dualrgb_v2.bag
Parsing description file /home/diogo/catkin_ws/src/robots/zau2/zau2_description/urdf/zau2.urdf.xacro
Executing command: xacro /home/diogo/catkin_ws/src/robots/zau2/zau2_description/urdf/zau2.urdf.xacro -o /tmp/description.urdf
Creating transformation tree using the urdf robot description ...
Running several verifications ... please wait ...
Checking if anchored sensor  is valid ... [OK]
Checking for compressed image topics ... 
Topic /camera/rgb/image_raw is in compressed format (/camera/rgb/image_raw/compressed). Will setup a decompressor.
Topic /camera_mb/rgb/image_raw is in compressed format (/camera_mb/rgb/image_raw/compressed). Will setup a decompressor.
Checking for throttled topics ... 
Sensor camera_ee throttled to 10 Hz.
Sensor camera_mb throttled to 10 Hz.
Checking for camera_info topic in bagfile ... [OK]
Checking if coordinate frames from config.yml exist in bagfile ... 
Unknown tag "dynamics" in /robot[@name='zau2']/link[@name='gripper_link']
Unknown tag "hardwareInterface" in /robot[@name='zau2']/transmission[@name='left_finger_trans']/actuator[@name='left_finger_motor']
Unknown tag "hardwareInterface" in /robot[@name='zau2']/transmission[@name='right_finger_trans']/actuator[@name='right_finger_motor']
Traceback (most recent call last):
  File "/home/diogo/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/scripts/configure_calibration_pkg", line 304, in <module>
    paths = nx.shortest_path(gx, config['world_link'], additional_tf['child_link'])
  File "/usr/local/lib/python3.8/dist-packages/networkx/classes/backends.py", line 145, in wrapper
    return func(*args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/networkx/algorithms/shortest_paths/generic.py", line 167, in shortest_path
    paths = nx.bidirectional_shortest_path(G, source, target)
  File "/usr/local/lib/python3.8/dist-packages/networkx/algorithms/shortest_paths/unweighted.py", line 221, in bidirectional_shortest_path
    raise nx.NodeNotFound(msg)
networkx.exception.NodeNotFound: Either source odom or target base_link is not in G

exit status: 1
```

and we found that the problem was that we should run the configure with the use tfs option.

Nonetheless, the error message is not very nice and should be improved.

miguelriemoliveira, 2023-10-18:
I was able to replicate. Instructions on how to in #638

miguelriemoliveira, 2023-10-18:
Here's a better print:
```
Checking for throttled topics ... 
Sensor camera_ee throttled to 10 Hz.
Sensor camera_mb throttled to 10 Hz.
Checking for camera_info topic in bagfile ... [OK]
Checking if coordinate frames from config.yml exist in bagfile ... 
ATOM Error: Additional tf for child link base_link not found.
Check the frames you entered in the config.yaml.
```
@Kazadhum , can you try it?

Kazadhum, 2023-10-18:
@miguelriemoliveira Yes! Just tried it and got the same print, which is much clearer than the traceback from NetworkX.

```
Checking for throttled topics ... 
Sensor camera_ee throttled to 10 Hz.
Sensor camera_mb throttled to 10 Hz.
Checking for camera_info topic in bagfile ... [OK]
Checking if coordinate frames from config.yml exist in bagfile ... 
ATOM Error: Additional tf for child link base_link not found.
Check the frames you entered in the config.yaml.
```

miguelriemoliveira, 2023-10-19:
OK, can you please add it? If not let me know and I will do it later.

Kazadhum, 2023-10-20:
Hey @miguelriemoliveira I forgot to reply in here but this is done. This is the new message:

```
ATOM Error: Additional tf for child link base_link not found.
Check the frames you entered in the config.yaml. Consider using the -utf/--use_tfs option in order to use the tranforms on the /tf and /tfs topics.
```

miguelriemoliveira, 2023-10-20:
Thanks. Looks great except the part 

on the /tf and /tfs topics

Should be

on the /tf and /tf_static topics

Kazadhum, 2023-10-20:
@miguelriemoliveira ah yes, my mistake. I'll fix it right away!

miguelriemoliveira, 2023-10-22:
Thanks @Kazadhum . Closing.

closed 2023-10-22 by miguelriemoliveira