lardemua/atom #655: MMTBot Dependencies

Kazadhum, 2023-10-22:
Hi @miguelriemoliveira! I was trying to launch the MMTBot system, but I ran into an error.

Running:
`roslaunch mmtbot_bringup bringup.launch  gui:=true`

results in:
```
RLException: while processing /home/diogo/catkin_ws/src/robots/mmtbot/mmtbot_gazebo/launch/mmtbot.launch:
while processing /home/diogo/catkin_ws/src/misc_pkgs/universal_robot/ur_gazebo/launch/mmtbot_ur10e_bringup.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: '/home/diogo/catkin_ws/src/misc_pkgs/universal_robot/ur_gazebo/launch/mmtbot_ur10e_bringup.launch'
The traceback for the exception was written to the log file
```

Looking `bringup.launch`, I noticed that it uses another launch file, `mmtbot.launch` (which is in the `mmtbot_gazebo` package). This one depends on the `ur_gazebo` package:

```
<include file="$(find ur_gazebo)/launch/mmtbot_ur10e_bringup.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="robot_description_file" value="$(find ur_gazebo)/launch/inc/load_mmtbot_ur10e.launch.xml"/>
        <!--    <arg name="robot_description_file" value="$(find ur_gazebo)/launch/inc/load_ur10e.launch.xml"/>-->
        <!--    <arg name="robot_description_file" value="$(find larcc_description)/urdf/larcc.urdf.xacro"/>-->
</include>
```

I assume that this `ur_gazebo` package is from the [universal_robot repository](https://github.com/ros-industrial/universal_robot). I don't think this repo is mentioned in the dependencies in the README for MMTBot.

However, another issue is that `ur_gazebo` doesn't have the file called in the line:

`<arg name="robot_description_file" value="$(find ur_gazebo)/launch/inc/load_mmtbot_ur10e.launch.xml"/>`

which results in my error.

manuelgitgomes, 2023-10-23:
Hello @Kazadhum!

> I assume that this `ur_gazebo` package is from the [universal_robot repository](https://github.com/ros-industrial/universal_robot).

That's why you should never assume anything :smile: 

Use this repo instead, in `calibration_devel` branch:

https://github.com/lardemua/universal_robot.git

miguelriemoliveira, 2023-10-23:
Thanks @manuelgitgomes ,

@Kazadhum , please add this note to the mmtbot installation section, if its not there.

Kazadhum, 2023-10-23:
Thank you @manuelgitgomes. Yes @miguelriemoliveira I'll do that as soon as possible.

miguelriemoliveira, 2023-10-27:
Is this solved? Should we close?

Kazadhum, 2023-10-27:
Hi @miguelriemoliveira I tried changing the readme but I can't push to the MMTBot repo... Could you add the line:

`git clone -b calibration_devel https://github.com/lardemua/universal_robot.git`

Other than that, we can close this issue, I think.

miguelriemoliveira, 2023-10-27:
Hi @Kazadhum ,

I added you as a collaborator of mmtbot. Please add the line

Kazadhum, 2023-10-27:
Thank you @miguelriemoliveira . By the way, I just remembered another issue with the dependencies that I ran into. I had to delete the `ur_kinematics` package from the `iris_ur10e` repo because it was repeated in the `lardemua/universal_robots` repo. I'll also point that out in the readme.

closed 2023-10-27 by miguelriemoliveira