lardemua/atom #656: Hand-eye evaluation

manuelgitgomes, 2023-10-23:
Hello @miguelriemoliveira!

To compare ATOM with other hand-eye calibration, some special metrics are needed.
They were already used on our paper "A General Approach to Hand–Eye Calibration Through the Optimization of Atomic Transformations".
Where are they? 
Should I create them?

miguelriemoliveira, 2023-10-23:
Hi @manuelgitgomes ,

I do not think there are any special metrics. If you read through the paper you will see we only use the reprojection error and the translation and rotation errors, exactly what we typically use in ATOM.

miguelriemoliveira, 2023-11-18:
Hi @manuelgitgomes ,

I think that for the reprojection errors we used two cameras. You are right, we did not have inter-collection errors at that time for sure.

closed 2023-11-18 by miguelriemoliveira