lardemua/atom #657: Problem with LiDAR - Depth evaluation

danifpdra, 2023-10-23:
Hi @miguelriemoliveira and @manuelgitgomes ,

I would like to get your opinion on this problem.

What's happening is that the LiDAR-depth pairs are giving high errors in the evaluation scripts. This is not due to bad calibration but a bug in the evaluation, or at least something we haven't thought about...

[image]

In this example, we have gaps in the depth limit points, but we have LiDAR points (blue) being projected to that area. What's happening is that our evaluation script is looking for the closest point and counting it, when these LiDAR points should not be considered for evaluation... 

Here are other examples 

[image]
[image]

In my opinion, we could either discard the points that are projected for NaN zones, or we could annotate borders in the depth image like we do for RGB to evaluate.... What do you think?

danifpdra, 2023-10-23:
In case you are wondering... These are the errors I am getting

[image]

For example collection 5 doesn't have a gap in the detection of the limits so the evaluation error is much lower

[image]

miguelriemoliveira, 2023-10-23:
Hi @danifpdra ,

first of all, congratulations of a very well written issue. I understood everything on the first read. You are improving a lot :+1: 

About the issue I would first say that this depth image is terrible. 

Then I would vote for the doing a manual annotation of the boundaries of the depth image. 
We can use the rgb manual annotation as a good starting point, but it does not have one functionality that we would need, which is to define only portions of the top line segment as valid. 

But I think it is easily adaptable, so if you want I can try to do it.

The other option of not evaluating lidar points which project to nan pixels in the depth image I don't like, because the location of the projection depends on the calibration ...

Lets hear what @manuelgitgomes has to say ...

danifpdra, 2023-10-24:
I agree with you @miguelriemoliveira. Yes, the depth image is horrible but that's most of the depth real images, unless you just have a RGB-D system and focus a lot of the pattern detection in depth. 

One problem that I see is that we would need to draw the rectangle but in some parts of the pattern we don't really know where it is. And if we cut out parts of the rectangle that we don't where it is in the annotation, we would still have the same problem of LiDAR points being projected to places where there is nothing. 

Example: 

I can easily draw a rectangle to fill these limits:

[image]

But in this example, I would only be guessing where the corners are, unless we have some sort of functionality that prolongs lines and allows us to find the intersection. 

[image]

manuelgitgomes, 2023-10-24:
Hello @danifpdra @miguelriemoliveira 

I agree with @miguelriemoliveira that a manual annotation can solve this, however, I believe this should be optional, as not all depth images need this.

Another option is to remove those images from the dataset with `-csf`

miguelriemoliveira, 2023-10-24:
> Another option is to remove those images from the dataset with `-csf`

That would be my favorite, but I don't think @danifpdra has enough "good images " to calibrate.

danifpdra, 2023-10-24:
> > Another option is to remove those images from the dataset with `-csf`
> 
> That would be my favorite, but I don't think @danifpdra has enough "good images " to calibrate.

What do you mean @miguelriemoliveira? This is evaluation, not calibration. I already calibrated.

miguelriemoliveira, 2023-10-24:
> What do you mean @miguelriemoliveira? This is evaluation, not calibration. I already calibrated.

Right. I meant enough good images to evaluate.

About your comments above, I think we can do an annotation that will not have a real depth measurement.

That means we would annotate the entire top, bottom, left and right lines of the pattern (the corners can be given as an intersection, but perhaps in a later stage).

After this, since we are computing the error by projecting lidar points into the depth image, its a good thing to have all the points along the boundaries of the pattern...

miguelriemoliveira, 2023-10-24:
Should I start?

miguelriemoliveira, 2023-10-26:
I will work on the annotation solution. Monitoring in #660

closed 2023-10-26 by miguelriemoliveira