lardemua/atom #658: Calibrate not deleting collections that do not have at least one RGB detection

danifpdra, 2023-10-24:
When the calibration script starts running, it outputs

`Deleted collections: []: at least one detection by a camera should be present.`

Nevertheless, it is not deleting collections where there isn't RGB detections. 

In this dataset, collection 19 doesn't have any RGB detection:

[image]

And the calibration script crashed with the following error 

`ValueError: Cannot set initial estimate for pattern at collection 019`

danifpdra, 2023-10-24:
FYI @miguelriemoliveira

miguelriemoliveira, 2023-10-26:
Hi @danifpdra ,

please give me the steps to reproduce so I can test a solution.

danifpdra, 2023-10-26:
Hi @miguelriemoliveira,

Collection 19 has no RGB detection. 

To calibrate without 19

`rosrun atom_calibration calibrate -json /home/daniela/datasets/larcc_subset/real/train/dataset_corrected.json -v -uic -csf 'lambda x: int(x) not in [19]'`

To calibrate with 19

`rosrun atom_calibration calibrate -json /home/daniela/datasets/larcc_subset/real/train/dataset_corrected.json -v -uic`

Dataset is: https://we.tl/t-4bHBXWmhdv
And should be inside datasets/larcc_subset/train/
Please pull larcc

miguelriemoliveira, 2023-10-27:
Hi @danifpdra ,

I looked into this. This is the expected behaviou, but the error message looked like a crash.

Now the output of the script it:

```terminal
ATOM Error: Cannot set initial estimate for pattern at collection 019. You must run the calibration without this collection. Check the csf flag.
```

That means that this collection does not have and rgb detection, and for that reason atom cannot create an initial estimate, which means it cannot carry out with the calibration.

closed 2023-10-27 by miguelriemoliveira