lardemua/atom #659: Evaluation from one camera 1 to camera 2 is much higher than the evaluation from camera 2 to camera 1

danifpdra, 2023-10-25:
Hi @miguelriemoliveira,

Do you know why when I run the RGB evaluation from one camera to another the values are much much higher than in the reverse? This makes no sense to me 

[image]

danifpdra, 2023-10-25:
BTW, the errors that came from the calibrate script are much much lower but the evaluations are still super high. I don't understand anything

miguelriemoliveira, 2023-10-25:
Hi @danifpdra ,

for me it will be difficult to look into this until the weekend.

Perhaps @manuelgitgomes has some insight ...

manuelgitgomes, 2023-10-25:
Hello @danifpdra and @miguelriemoliveira 

> the errors that came from the calibrate script are much much lower but the evaluations are still super high

This seems to me like the optimization process landed on a local minimum.
Visualize the results of the calibration.
If something seems wrong, maybe try another initial estimate.

> Do you know why the evaluation is so different from one dataset to another when the datasets are from the same bag? This makes no sense whatsoever to me

Regarding this, try and visualize the results of the evaluation.
If the system is indeed on a local minimum, one dataset might favor that minimum when compared to another.

danifpdra, 2023-10-25:
Hi @miguelriemoliveira,

I already did that and the results seem ok to me... 
[image]

miguelriemoliveira, 2023-10-26:
Hi @danifpdra ,

I am wondering about the tables printed by each calibration during calibration. Can you put them here? 
That could give us a hint if the problem is related  to the evaluation or something more...

danifpdra, 2023-10-26:
> > BTW, the errors that came from the calibrate script are much much lower but the evaluations are still super high. I don't understand anything
> 
> Wait, now I am reading this and this kind of answers my question. So that means the problem is only in the evaluation?

I don't think so. When I calibrate just the two RGB cameras the evaluation still gives me like 10px of error. But I know the methodology works because I calibrated the simulated system and it works flawlessly

miguelriemoliveira, 2023-10-26:
> I don't think so. When I calibrate just the two RGB cameras the evaluation still gives me like 10px of error. But I know the methodology works because I calibrated the simulated system and it works flawlessly

I am confused, this points to the problem being in the evaluation, right? Am I misunderstanding?

danifpdra, 2023-10-26:
I don't think so. The evaluation works well with the simulated dataset

miguelriemoliveira, 2023-10-27:
Hum, but it can be something in the evaluation that only occurs with real data.

When you say: "But I know the methodology works because I calibrated the simulated system and it works flawlessly"
To me it seems that the calibration is working fine (I guess you see the error tables printed during calibration and they are fine), and the problem is in the evaluation. I am still lost.

Where do you think the problem is?

danifpdra, 2023-10-27:
I honestly don't know... Let's take this example evaluating the hand_camera_depth to rgb_camera_3 pairs.

Collection 000; RMS 5.32 - Seems fine

[image]

Collection 016; RMS 9.46 - seems displaced

[image]

Collection 017; RMS 12.33

[image]

Evaluation seems to be matching the images, I think, but the errors are very high sometimes. In the same dataset I have RMS varying from 25 to 1 in difference collections 

And then in the RGB evaluation, all the values are very high and the evaluation is 57px...

[image]

manuelgitgomes, 2023-10-27:
> Hum, but it can be something in the evaluation that only occurs with real data.

This seems unlikely to me, at least for the rgb to rgb evaluation, which has been the same since the beginning.

Does this only occur for sensors in the hand? Is the hand RGB-D corrected?

danifpdra, 2023-10-27:
> > Hum, but it can be something in the evaluation that only occurs with real data.
> 
> This seems unlikely to me, at least for the rgb to rgb evaluation, which has been the same since the beginning.
> 
> Does this only occur for sensors in the hand? Is the hand RGB-D corrected?

yes, one RGB and one depth in the hand. what do you mean corrected?

manuelgitgomes, 2023-10-27:
> yes, one RGB and one depth in the hand. what do you mean corrected?

The problem we had with RGB-D in the gantry. We changes the xacros to correct it.
Is it applied to the hand RGB-D?

danifpdra, 2023-10-27:
> > yes, one RGB and one depth in the hand. what do you mean corrected?
> 
> The problem we had with RGB-D in the gantry. We changes the xacros to correct it. Is it applied to the hand RGB-D?

I didn't know of any problem, but I used the set_initial_estimate to improve the first guess so it would eliminate that, no?

miguelriemoliveira, 2023-10-27:
Hi @danifpdra ,

I think I may have found the problem. When I refactored the pattern detection code some weeks ago I changed the coordinate system of the pattern.

[image]

Now the generation of the ground truth boundary  points used to compute the cost function is based on this coordinate frame. 

You can see in the image that the points are not well positioned, and I think that's the problem. I think all range sensors work fine because the error is like a bias  for all sensors, but in the case of the rgb sensor there is a clear inconsistency which should generate the high errors you refer.

I have no clue why it worked on simulation tough ...

I will continue to investigate.

miguelriemoliveira, 2023-10-27:
Fixed the generation of ground truth data:

[image]

I will now try to evaluate the impact ...

miguelriemoliveira, 2023-10-28:
Continuing on #663

closed 2023-10-28 by miguelriemoliveira