lardemua/atom #661: Use depth annotations in lidar_to_depth_evaluation

miguelriemoliveira, 2023-10-26:
After #660, we should use the depth annotations in the evaluation scripts.

miguelriemoliveira, 2023-10-26:
I think its working. 

[image]

@danifpdra , can you try?

First annotate the depth sensor depth images using the **annotate_pattern_borders_in_rgb_or_depth**.

Then, use the **lidar_to_depth_evaluation** script with the --use-annotations flag.

danifpdra, 2023-10-27:
The script works fine, thank you

closed 2023-10-27 by miguelriemoliveira

miguelriemoliveira, 2023-10-28:
> The script works fine, thank you

781433 + 1 coffees

danifpdra, 2023-10-28:
> > The script works fine, thank you
> 
> 781433 + 1 coffees

Maybe I should start counting coffees for all the bugs I've found in ATOM......

miguelriemoliveira, 2023-10-28:
Not sure what you are talking about ...

danifpdra, 2023-10-28:
yes, yes... you should give me a position as ATOM's quality control engineer