lardemua/atom #663: Fix pattern coordinate system inconsistencies across modalities

miguelriemoliveira, 2023-10-28:
On the sequence of #659 and the recent change of the charuco detector which imposed a change in the location of the pattern coordinate system (#630 ).

Also related to #633

miguelriemoliveira, 2023-10-28:
I think I got it. It had to do with the opencv changes.

The old way of computing the pattern's coordinate system (solvepnp) would find the pattern assuming the coordinate system to be in the top left corner. 

The new way of computing the pattern's pose assumes (cv2.aruco.estimatePoseCharucoBoard) assumes the coordinate frame to be on the upper left corner of the board.

See the image for the differences:

[image]

miguelriemoliveira, 2023-10-28:
So now I think everything is fixed properly. The coordinate system is always the top left corner of the board (the one which is higher and more to the left in the image).

I had to fix also the use of the cv2.solvepnp, which uses a different convention, and the generation of ground truth data for using in the objective function.

Now I think everything is working fine, but I will make some tests and add them here.

miguelriemoliveira, 2023-10-28:
**Test 1** , for the rgb_depth_system, when starting with a nig of 0 0, the initial values should be close to perfect and they are

[image]

and also, the visualization is fine:

[image]

miguelriemoliveira, 2023-10-28:
**Test 2**, for the rgb_rgb_system, the calibration starts with an almost perfect error (because we use nig 0 0)

[image]

and the visualization is also consistent

[image]

miguelriemoliveira, 2023-10-28:
@manuelgitgomes , @danifpdra and @Kazadhum ,

I am going to push these changes to atom's main branch because the previous version had several bugs. 

The problem is that the calibrations that were carried out since the change in the convention of the pattern's coordinate system until now are invalid.

That was tree weeks ago (https://github.com/lardemua/atom/issues/630#issuecomment-1751774307)

I do not think that the problem affected the generation of bagfiles, nor the datasets.

**But you should discard any calibrations you did during this period**. I am really sorry for this ...

This should explain the bugs described in #659 and also perhaps some problems @Kazadhum was having with the calibrations (do not know the issue).

miguelriemoliveira, 2023-10-28:
Just merged https://github.com/lardemua/atom/issues/630#issuecomment-1751774307

danifpdra, 2023-10-28:
I will test on the system that wasn't working and get back to you

danifpdra, 2023-10-28:
> @manuelgitgomes , @danifpdra and @Kazadhum ,
> 
> I am going to push these changes to atom's main branch because the previous version had several bugs.
> 
> The problem is that the calibrations that were carried out since the change in the convention of the pattern's coordinate system until now are invalid.
> 
> That was tree weeks ago ([#630 (comment)](https://github.com/lardemua/atom/issues/630#issuecomment-1751774307))
> 
> I do not think that the problem affected the generation of bagfiles, nor the datasets.
> 
> **But you should discard any calibrations you did during this period**. I am really sorry for this ...
> 
> This should explain the bugs described in #659 and also perhaps some problems @Kazadhum was having with the calibrations (do not know the issue).

Don't you think this will affect the collecting of RGB data? This didn't affect the labeling of RGB? Just the calibration?

miguelriemoliveira, 2023-10-28:
Now I am running the evaluation of the rgb_depth_system and I am getting 9 pixels of error on average, which is worse than expected. I will try with the rgb_rgb_system and get back to you.

miguelriemoliveira, 2023-10-28:
> Don't you think this will affect the collecting of RGB data? This didn't affect the labeling of RGB? Just the calibration?

I think not. The labeling of rgb is done in two stages. 

1. Finding the corners, which is done during collection

https://github.com/lardemua/atom/blob/f5ad1e70a9d35b170a0ec6087b8eb77ee33fcfe3/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py#L390-L411

but this only stores pixel coordinates, so the change does not affect it.

2. Estimating the pattern 3d pose from the stored corners, this is done during calbration and is what is affected by the change

https://github.com/lardemua/atom/blob/f5ad1e70a9d35b170a0ec6087b8eb77ee33fcfe3/atom_calibration/src/atom_calibration/calibration/patterns_config.py#L190-L238

So I think the collection of datasets was not affected.

danifpdra, 2023-10-28:
It's muhc better but still high...

[image]

[image]

miguelriemoliveira, 2023-10-28:
oops , reopening ... because evaluation is not running well

miguelriemoliveira, 2023-10-28:
I also get these errors in the rgb_depth_system, depth_to_rgb_evaluation script

```terminal
Errors per collection
+--------------+-----------+-------------+-------------+
| Collection # | RMS (pix) | X err (pix) | Y err (pix) |
+--------------+-----------+-------------+-------------+
|     000      |  10.2982  |    6.7706   |    7.3185   |
|     001      |   9.9239  |    6.3307   |    6.5826   |
|     002      |   9.1868  |    6.4452   |    6.0405   |
|     003      |   8.9215  |    5.8380   |    6.1968   |
|   Averages   |   9.5826  |    6.3461   |    6.5346   |
+--------------+-----------+-------------+-------------+
```

miguelriemoliveira, 2023-10-28:
and I get these strange projections where the pattern looks smaller

[image]

[image]

miguelriemoliveira, 2023-10-28:
Did an experiment with the rgb_rgb_system and the calibration went great, and then the evaluation gave the results

[image]

which are great as well. So it seems that the problem is not on the rgb part, perhaps the depth_to_rgb_evaluation script ...

miguelriemoliveira, 2023-10-28:
Going back to the rgb_depth_system, the calibration visually looks good,

relative pose between sensors goes from this:

[image]

to this:

[image]

miguelriemoliveira, 2023-10-28:
Did another calibration with nig 0 0, same behavious. Perfect calibration, evaluation with 10 pixels error on average

miguelriemoliveira, 2023-10-28:
@danifpdra , did you ever get very good results in the **depth_to_rgb_evaluaion**? Did you ever get 1 pixel or below on average?

danifpdra, 2023-10-28:
I did have results around 1 pixel. You can see the tables in my paper on overleaf but I'm not sure if the calibration wasn't before you messed with the chessboard coordinate frame...

miguelriemoliveira, 2023-10-28:
Thanks I will try to continue tonight ...

miguelriemoliveira, 2023-10-28:
Got it. This is wrong

https://github.com/lardemua/atom/blob/dec5479fb7caf72bea088a03c4921889852fdcdc/atom_evaluation/scripts/depth_to_rgb_evaluation#L107

Because we are projecting the **boundary points in the depth image of the train collection**, into **the rgb image of the test collection**. That does not make sense.

I think the best is to also use depth annotations and project those from the depth image of the test collection, into the rgb image of the test collection.

Will implement in #669

danifpdra, 2023-10-28:
But the problem I was having was just in the rgb pairs...

miguelriemoliveira, 2023-10-28:
> Got it. This is wrong
>
> https://github.com/lardemua/atom/blob/dec5479fb7caf72bea088a03c4921889852fdcdc/atom_evaluation/scripts/depth_to_rgb_evaluation#L107
>
> Because we are projecting the **boundary [...]

Made a mistake here as @danifpdra told me. This reasoning is wrong.

miguelriemoliveira, 2023-10-29:
Closing the issue. I think the problem of the pattern coordinate frame conventions is solved ...

closed 2023-10-29 by miguelriemoliveira