lardemua/atom #666: Improve names of evaluations in full evaluations

miguelriemoliveira, 2023-10-28:
Hi @manuelgitgomes ,

I was using the full_evaluation script and got confused with one thing. The print said:

    process[rgb_to_depth_evaluation-2]: started with pid [15409]

which was confusing since the actual script name is  **depth_to_rgb_evaluation**.

I changed the template from this:

```xml
{% for combination in c.sensor_combinations -%}

    <node type="{{ c.sensor_combinations[combination]['filename'] }}_evaluation" pkg="atom_evaluation" 
        name="{{ combination }}_evaluation"
        args="{{ c.sensor_combinations[combination]['args'] }} -train_json $(arg train_json) -test_json $(arg test_json) -csf '$(arg csf)' -sfr" 
        output="screen"/>
```

to this:

```xml
{% for combination in c.sensor_combinations -%}

    <node type="{{ c.sensor_combinations[combination]['filename'] }}_evaluation" pkg="atom_evaluation" 
        name="{{ c.sensor_combinations[combination]['filename'] }}_evaluation"
        args="{{ c.sensor_combinations[combination]['args'] }} -train_json $(arg train_json) -test_json $(arg test_json) -csf '$(arg csf)' -sfr" 
        output="screen"/>
```

and it seems to work fine. I pushed, please close if you agree, or let me know if I did something wrong.

manuelgitgomes, 2023-10-30:
Hello @miguelriemoliveira.
This will lead to problems when running a large system with many sensor. 
Imagine we have 3 RGB cameras, we will have to do 3 RGB to RGB evaluations. 
However, they will all have the name `rgb_to_rgb_evaluation`, which will lead to errors.

What was present before your commit was my solution for this, but it lead to your misunderstanding.
It might have been more confusing because your sensors are named `rgb` and `depth`.
If they were named `rgb_camera` and `depth_camera`, the node would be named `rgb_camera_to_depth_camera`, which is still wrong, but less confusing.
We can find a way to specify the order of the names (`depth_to_rgb` rather than `rgb_to_depth`)

manuelgitgomes, 2023-11-22:
Sorry for the late response @miguelriemoliveira.

> Then we say <sensor1_name> <sensor2_name> evaluation

This already happened before the change.
The thing is that it does not check for the correct order.

manuelgitgomes, 2023-11-22:
Done!

This was done by verbosely explaining every case so it can be correctly named.

closed 2023-11-22 by manuelgitgomes

miguelriemoliveira, 2023-11-22:
Great! Thanks.