lardemua/atom #667: Zau2 calibration configuration error

Kazadhum, 2023-10-28:
Hi @miguelriemoliveira and @manuelgitgomes! I finally managed to build `zau` from scratch using `mmtbot`, as you suggested in our previous meeting. MoveIt is working and so is the robot's teleop.

I recorded a bagfile, which you can download [here](https://we.tl/t-ODZ45qXmLv) and I tried configuring the calibration package. However, I got an error:

```
Configuring calibration package zau2_calibration
_______________________________________________________

Loading config_file /home/diogo/catkin_ws/src/robots/zau2/zau2_calibration/calibration/config.yml
Loading bagfile /home/diogo/bagfiles/zau2_mmtzau/bag1.bag
Parsing description file /home/diogo/catkin_ws/src/robots/zau2/zau2_description/urdf/zau2.urdf.xacro
Executing command: xacro /home/diogo/catkin_ws/src/robots/zau2/zau2_description/urdf/zau2.urdf.xacro -o /tmp/description.urdf
Creating transformation tree using the urdf robot description ...
Running several verifications ... please wait ...
Checking if anchored sensor  is valid ... [OK]
Checking for compressed image topics ... 
Topic /hand_camera/rgb/image_raw is in compressed format (/hand_camera/rgb/image_raw/compressed). Will setup a decompressor.
Topic /camera_mb/rgb/image_raw is in compressed format (/camera_mb/rgb/image_raw/compressed). Will setup a decompressor.
Checking for throttled topics ... 
Sensor hand_camera throttled to 10 Hz.
Sensor camera_mb throttled to 10 Hz.
Checking for camera_info topic in bagfile ... [OK]
Checking if coordinate frames from config.yml exist in bagfile ... 
ATOM Error: Additional tf for child link base_link not found.
Check the frames you entered in the config.yaml. Consider using the -utf/--use_tfs option to use the tranforms on the /tf and /tf_static topics.
```

I am using `-utf` and the `base_link` link is present in the transformation tree (picture below):

[image]

In order to replicate, make sure you have `mmtbot` working in your system and clone the `mmtzau` branch of the `zau2` repo ([link here](https://github.com/Kazadhum/zau2/tree/mmtzau)). Download the bag file to your ROS_BAGS directory and change the path to the bag file in `config.yml` accordingly.

Do you have any insight or tips for debugging?

miguelriemoliveira, 2023-10-28:
> I finally managed to build `zau` from scratch using `mmtbot`, as you suggested in our previous meeting.

Great news.

About the problem. The configuration file has this additional tf

https://github.com/Kazadhum/zau2/blob/d225a50cd347512b23ac84e531a0676c1490352f/zau2_calibration/calibration/config.yml#L93-L100

and looking and the tf tree picture you posted I can find both **base_link_mb** as well as **base_link**.

Then I wondered if the bagfile actually contains these transforms

So I played the bagfile and ran

    rostopic echo /tf | grep "child_frame_id: \"base_link\"" -A 2 -B 2 

which printed nothing, which means that a transformation where child_frame_id is "base_link" does not exist in the bagfile ...

not in the tf topic anyway ... so I searched for the same transform in the tf_static topic

    rostopic echo /tf_static | grep "child_frame_id: \"base_link\"" -A 2 -B 2

and got the print

```terminal
 frame_id: "base_link_mb"
    child_frame_id: "base_link"
    transform: 
      translation: 
```

So the transform exists in the bagfile, but the configure script is launching an error ... 

this needs further investigation. I will get back to you ...

miguelriemoliveira, 2023-10-28:
I can replicate the error ...

miguelriemoliveira, 2023-10-28:
I am [plotting the gx graph](https://github.com/lardemua/atom/blob/f5ad1e70a9d35b170a0ec6087b8eb77ee33fcfe3/atom_calibration/scripts/configure_calibration_pkg#L202-L203) which is used to search for the base_link frame, and I get this:

[image]

in the middle there is a base_link node ... so I am still not understanding what's happening ...

miguelriemoliveira, 2023-10-28:
I found the problem. This code line:

https://github.com/lardemua/atom/blob/f5ad1e70a9d35b170a0ec6087b8eb77ee33fcfe3/atom_calibration/scripts/configure_calibration_pkg#L305-L307

does not make sense. It must be **if not** to test if the frame does not exist.

I wonder how this worked before ... I will look for changes in the file ...

miguelriemoliveira, 2023-10-28:
It was ok here 

https://github.com/lardemua/atom/commit/7d1c86b758f4be610d66aff4a9ac4b2e491b8315#diff-933a3af1b770e462dfcccd4b5243bb44aaee43386fde2943dd3dc2934dbd72f3R281

miguelriemoliveira, 2023-10-28:
Searched around but could not find which commit changed this test. Anyway, it should be working now.

miguelriemoliveira, 2023-10-28:
@Kazadhum , I have another error further down the road

```
ATOM Error: Frame world given in config file for calibration_pattern.pattern_link does not exist. Check config file /home/mike/workspaces/catkin_ws/src/calibration/robots/zau2/zau2_calibration/calibration/config.yml
```

but I think that one you can solve easily ...

closed 2023-10-28 by miguelriemoliveira

Kazadhum, 2023-10-28:
Got it @miguelriemoliveira ! Thank you! It worked now.

commit dec5479f (2023-10-28): Fix #667