lardemua/atom #670: Labeller not working during data collection

Kazadhum, 2023-10-28:
When running:

`roslaunch zau2_calibration collect_data.launch output_folder:=$ATOM_DATASETS/zau2_mmtzau/dataset1 overwrite:=true`

I get the following error message:

```
[ERROR] [1698527273.996925, 559.465157]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7f68e75bcb50>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 264, in sensorDataReceivedCallback
    self.labelData()  # label the data
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 391, in labelData
    result = self.pattern.detect(image, equalize_histogram=False)
  File "/home/diogo/catkin_ws/src/atom/atom_calibration/src/atom_calibration/collect/patterns.py", line 120, in detect
    ret, ccorners, cids = cv2.aruco.interpolateCornersCharuco(corners, ids, gray, self.board)
AttributeError: module 'cv2.aruco' has no attribute 'interpolateCornersCharuco'
```

and in RViz:
[image]

If you want to try to replicate, clone the [zau2 robotic system repo](https://github.com/Kazadhum/zau2/tree/mmtzau) (use the `mmtzau` branch) and make sure you have `mmtbot` working on system.

I'll see if I can figure this out tonight, since it seems to be an issue with cv2, but I wonder why it works for everyone else and for other systems as well.

Tagging @miguelriemoliveira and @manuelgitgomes for visibility.

Kazadhum, 2023-10-28:
I'll test the data collection with the rgb_rgb_system from the ATOM examples before trying to mess with cv2

Kazadhum, 2023-10-28:
Ok so the same thing happens when I try to run `collect_data` in the `rgb_rgb_system`...

miguelriemoliveira, 2023-10-28:
Hi @Kazadhum ,

this sounds like an opencv version problem. What is your version?

miguelriemoliveira, 2023-10-28:
Sounds similar to this

https://github.com/lardemua/atom/issues/629#issuecomment-1752603114

Kazadhum, 2023-10-28:
Hi @miguelriemoliveira! I just fixed it by installing:
 
`pip install -U opencv-contrib-python`

So it turned out it was an opencv version problem. Now I have version 4.8.1.

miguelriemoliveira, 2023-10-28:
Great. Closing issue then.

miguelriemoliveira, 2023-10-28:
> `pip install -U opencv-contrib-python`
> 
> So it turned out it was an opencv version problem. Now I have version 4.8.1.

In any case this solution sounds strange because the charuco moved from the opencv-contrib into the opencv branch, so in theory you shouldn't even need the opencv-contrib. 

In any case I would move along ... until the next bug ...

closed 2023-10-28 by miguelriemoliveira

manuelgitgomes, 2023-10-30:
> In any case this solution sounds strange because the charuco moved from the opencv-contrib into the opencv branch, so in theory you shouldn't even need the opencv-contrib.

I believe opencv-contrib is a superset of opencv, so by updating it, @Kazadhum updated opencv as well.

miguelriemoliveira, 2023-10-30:
Nice, now I understand.