lardemua/atom #671: Fixing rgb to rgb evaluation

danifpdra, 2023-10-28:
RGB to RGB evaluation was wrong

When we evaluate a calibration, we need two datasets: a train dataset, where we get the transformations between the links and a test dataset, where we get the data to evaluate. 

The problem is that in this script, we were evaluating were using the test dataset transformations to evaluate the script, which caused big problems specially with real data... I fixed this and will investigate further on other scripts to make sure this didn't happen in others.

danifpdra, 2023-10-28:
FIY @manuelgitgomes @miguelriemoliveira

miguelriemoliveira, 2023-10-28:
This was a great catch @danifpdra . Congratulations!

miguelriemoliveira, 2023-10-29:
@danifpdra , don't forget to check if all other evaluation script have a similar problem.

Thanks again for finding this big bug.

danifpdra, 2023-10-29:
Getting weirdly high translation errors for real data... will investigate

[image]

miguelriemoliveira, 2023-11-14:
Closing this one.

closed 2023-11-14 by miguelriemoliveira