lardemua/atom #672: Simple control experiment rgb_depth_system does not give expected results

miguelriemoliveira, 2023-10-29:
Tagging @danifpdra , @manuelgitgomes , @Kazadhum for visibility.

Working on #642 we get a final depth to rgb evaluation of 2 pixels of error. This does not seem bad, but since this is a simulated system I wounder if we should be having more. Also, @danifpdra reports getting evaluations of this type around 1 pixel, so I am creating this issue to investigate.

The experiment is very simple.

   - 1 collection only
   -  using as train the **train dataset dataset_corrected.json**, which contains initial transformation values, which are perfect since this is a simulation 
   -  carry out evaluation using:
       - as train use the **train dataset dataset_corrected.json**
       - as test the same **train dataset dataset_corrected.json**

This means that we are "pseudo" calibrating for a single collection, and have the perfect ground truth transformation. 
Then we evaluate on a test dataset which is actually the same as used for training.

The error in this case should be close to zero, right? Am I making any mistake?

miguelriemoliveira, 2023-10-29:
The results are:

```terminal
Errors per collection
+--------------+-----------+-------------+-------------+
| Collection # | RMS (pix) | X err (pix) | Y err (pix) |
+--------------+-----------+-------------+-------------+
|     000      |   4.5850  |    2.1880   |    2.2055   |
|   Averages   |   4.5850  |    2.1880   |    2.2055   |
+--------------+-----------+-------------+-------------+
```

this is very far from a perfect result. The image shows:

[image]

it seems the error is only on the bottom ...

[image]

miguelriemoliveira, 2023-10-29:
The labels for the depth image in fact seem to be wrong in the bottom

[image]

miguelriemoliveira, 2023-10-29:
I improved the depth boundary labeling 

[image]

and rerun the evaluation. I get this:

[image]

which is much better.

miguelriemoliveira, 2023-10-29:
With a better visualization we have:

[image]

miguelriemoliveira, 2023-10-29:
I think there might be a bug here ... doing a preposterous annotation

[image]

gets this

[image]

I am not sure this is correct. Will think about it later ...

danifpdra, 2023-10-29:
Hi @miguelriemoliveira,

Why do you think that's not correct? That seems correct to me

miguelriemoliveira, 2023-10-29:
Here's a better visualization:

Green -> the annotated points
red -> the projected points
yellow -> the line found from each projected point to the closest annotated point

[image]

miguelriemoliveira, 2023-10-29:
Yeah, it looks good. So if I annotate the bottom line a bit to the inside of the pattern

[image]

we get this:

[image]

which results in table:

[image]

I think I am convinced that I am sure that I am confident that these errors are a natural part of the process, and not because of some bug. 

@danifpdra , are you also convinced that you are sure that you are confident?

danifpdra, 2023-10-29:
> Yeah, it looks good. So if I annotate the bottom line a bit to the inside of the pattern
> 
> [image]
> 
> we get this:
> 
> [image]
> 
> which results in table:
> 
> [image]
> 
> I think I am convinced that I am sure that I am confident that these errors are a natural part of the process, and not because of some bug.
> 
> @danifpdra , are you also convinced that you are sure that you are confident?
> 
> Should I stop investigating this? I think I should ...

I am absolutely convinced that I'm sure that I'm confident, yes.

I reran all my experiments from camerabot (RGB-D) real and simulated and here's my results. It all seems good to me 

[image]

miguelriemoliveira, 2023-10-29:
OK, thanks @danifpdra . Closing the issue.

closed 2023-10-29 by miguelriemoliveira