lardemua/atom #675: Is RGB calibration really right?

danifpdra, 2023-11-08:
I am getting weird calibration output errors for RGB pairs. Check more details on this issue https://github.com/danifpdra/rgbdbot/issues/2

danifpdra, 2023-11-08:
I am using a chessboard instead of a charuco... Could this be the problem? 

https://github.com/lardemua/atom/blob/8498653f755f7900e7b89dae35b7c638524bffb1/atom_calibration/src/atom_calibration/calibration/patterns_config.py#L230-L239

miguelriemoliveira, 2023-11-08:
Hum ... I think this is the problem I solved for the new opencv version in the rgb_labeler. 
Perhaps the problem is still there on the opencv calibration script

I will take a look tonight and try to call...

danifpdra, 2023-11-08:
Ok, let me know then

miguelriemoliveira, 2023-11-08:
Hi @danifpdra ,

Sorry, only now I am free and too sleepy to work. Lets talk tomorrow afternoon, ok?

Again sorry,
Miguel

miguelriemoliveira, 2023-11-09:
Lets work on #654 to see if chessboards work fine.

miguelriemoliveira, 2023-11-14:
@danifpdra can we close this?

miguelriemoliveira, 2023-11-18:
Closing for now.

closed 2023-11-18 by miguelriemoliveira