lardemua/atom #676: Place pattern link in calibration configuration transformation tree PDF

Kazadhum, 2023-11-08:
What the title says. I think it would be useful to spot mistakes in the calibration configuration that wouldn't cause an error when running the script (such as poorly defined pattern parent links).

Tagging @miguelriemoliveira, @manuelgitgomes and @danifpdra for visibility.

miguelriemoliveira, 2023-11-15:
Here's an example of what we have now.

[image]

The goal is to also represent the calibration pattern and the transformation from it to its parent coordinate frame.

miguelriemoliveira, 2023-11-15:
@Kazadhum , when creating new issues always add them to project Ongoing work on general calibration (the only existing one) please.

miguelriemoliveira, 2023-11-15:
Implemented. Here's the new look:

[image]

also works for additional_tfs, here's an example with an additional transform from the tripod to the tripod_plate:

[image]

closed 2023-11-15 by miguelriemoliveira

Kazadhum, 2023-11-15:
> @Kazadhum , when creating new issues always add them to project Ongoing work on general calibration (the only existing one) please.

@miguelriemoliveira will do!

PR #685 (2023-11-15): Close #676. Implemented.
Close #676
commit 193703d5 (2023-11-15): Close #676. Implemented.
commit 29076b30 (2023-11-15): Merge pull request #685 from lardemua:miguelriemoliveira/issue676

Close #676. Implemented.