lardemua/atom #677: Possibility of using multiple calibration patterns during calibration

manuelgitgomes, 2023-11-09:
Hello @miguelriemoliveira.
More of a thought experiment for now, but is it theoretically possible to have multiple calibration patterns during an ATOM calibration?

When calibrating the PR2, it would be interesting to have a calibration pattern w.r.t. the world frame and one (or two) w.r.t. the end effector(s) of the robot. 
This would allow for a eye-to-base calibration to be done simultaneously with a "regular" calibration, allowing for the calibration of the sensors w.r.t. the robot frame and w.r.t. the end effector.
The two (or three) calibrations would need to be done, be it separately (as they currently are), or simultaneously, which has never been done to my knowledge.
I think that doing it simultaneously might even create better estimations, as every detection is used in a single cost function.

I know that, in practice, a lot of code would need to be changed, but I am just asking to access if this is a topic that merits a dive in.

miguelriemoliveira, 2023-11-09:
Hi @manuelgitgomes ,

nice idea. I think its possible in theory. We would have to use identifiable patterns, but I think we can have different charuco patterns with different charuco dictionaries. So its possible to detect two or more patterns without confusing them.

From here on onward I think we could with relative ease (2 to 3 weeks) extend atom to work with more than one pattern.

In fact, the **sensor to pattern** approach of atom makes it more or less straigthforward.

In the **sensor to pattern** approach we compute for each sensor an error by projecting the pattern to the sensor.

In the novel **sensor to pattern(s)** we would have one error per each sensor pattern combination, and send all that into the optimizer. 

So I think yes, its possible, and I am already having a glimpse on how it could be done.

About the interest, I like the idea, and I think it would be publishable. 

What does @v4hn say about it?

miguelriemoliveira, 2023-11-18:
Closing this one. For now it does not seem to be a priority.

closed 2023-11-18 by miguelriemoliveira