lardemua/atom #679: Use of XACRO to retrieve robot name

manuelgitgomes, 2023-11-13:
Hello @miguelriemoliveira,

In this block of description_file, _, _ = atom_core.config_io.uriReader(self.config['description_file'])
        description = URDF.from_xml_file(description_file)

miguelriemoliveira, 2023-11-13:
Hi @manuelgitgomes ,

thanks for the comments. I understand the limitations you point out. They make sense.

I do not like your suggestion because it brakes all datasets we have collected before.
Also, couldn't we retrieve the name of the robot by parsing the xacro when configuring the calibration package?

manuelgitgomes, 2023-11-14:
Hello @miguelriemoliveira,

Yes, good point.
We can see if there is a python library that parses it, if not, we can convert the xacro to urdf beforehand and parse the urdf

miguelriemoliveira, 2023-11-14:
Hi @manuelgitgomes ,

My suggestion would be to convert the xacro to an urdf and put that urdf in the /tmp folder. 
Then analyse the urdf. 

We do this already somewhere in ATOM.

manuelgitgomes, 2023-11-29:
Closed with #682

closed 2023-11-29 by manuelgitgomes