lardemua/atom #680: ATOM assumes calibration package name

manuelgitgomes, 2023-11-13:
In this line:

https://github.com/lardemua/atom/blob/b673399dd6850a0c4ac2f7f53a2177274b06c8d6/atom_core/src/atom_core/xacro_io.py#L82

ATOM assumes that the calibration package is `name_of_the_robot_calibration`, which is a good practice, but it might not be the case for any number of reasons.

The name of the calibration package should be in the metadata as well.

miguelriemoliveira, 2023-11-13:
Right, we can add this directly on configure time. I will try to do it.

Kazadhum, 2023-11-14:
Hello! I think this is the same issue as #653! So if you close this one I the other one should be closed as well, I think.

manuelgitgomes, 2023-11-14:
Thanks @Kazadhum, I will close this and continue the discussion in #653

closed 2023-11-14 by manuelgitgomes