lardemua/atom #681: playbag.launch has wrong path to optimized URDF

manuelgitgomes, 2023-11-13:
In the playback template:

https://github.com/lardemua/atom/blob/b673399dd6850a0c4ac2f7f53a2177274b06c8d6/atom_calibration/templates/playbag.launch#L42-L43

The optimized path should be `package_name/urdf/optimized/optimized.urdf.xacro`

miguelriemoliveira, 2023-11-13:
I remember @danifpdra was doing something here, but I never like the solution much. @danifpdra , can you explain please?

danifpdra, 2023-11-14:
Hummm that is wrong because the file was being saved in the wrong folder... later on I fixed that but I must've not change the playbag.launch...

miguelriemoliveira, 2023-11-14:
Ok, thanks @danifpdra . Will you fix it then?

miguelriemoliveira, 2023-11-19:
Fixed. Now we can calibrate and after we run 

roslaunch rgb_hand_eye_system_calibration playbag.launch optimized:=true

and can see how well things overlay. For example, running without a good calibration, we get:

[image]

and after a good calibration the overlay is perfect.

[image]

closed 2023-11-19 by miguelriemoliveira