lardemua/atom #687: Automatic data collection

manuelgitgomes, 2023-11-16:
I think that having an automatic data collection script would be very useful.

My case is that I have a very large bagfile (almost 1 hour), and to select the collections by hand is very cumbersome.

One simple way to do this is to read the bag file and take collections when no tf has changed in x seconds.
This does not cover the case where the robotic system is fixed and the pattern is changing.
To cover this, we can check the corner detections on rgb sensors and see if they have moved.

For this, it would also be useful to add the functionality to fully remove collections from the dataset, as some might be wrongfully taken.
I suggest that this should be added to the `inspect_atom_dataset` (even fusing with `rename_atom_dataset` and having a general dataset changing script)

miguelriemoliveira, 2023-11-16:
Hi @manuelgitgomes ,

thanks for the idea. It makes sense, and I like your ideas for how to automatically filter the collections. 

I know kalibr and opencv do this automatic collection of data.

I also see a paper comming where we can evaluate the impact of autocollecting datasets by comparing the results of calibration using manual vs auto collection of data (with different flavors or filtering).

With @Kazadhum we already did something similar, but which was not fully general. But we were aiming at producing the bagfile. Having a bagfile and extracting collections should be easier, and perhaps a better idea.

So in general I really like the idea. Can I try to do it over the weekend or you want to do it yourself?

Kazadhum, 2023-11-16:
Hi @miguelriemoliveira and @manuelgitgomes !

This sounds like a great idea! While working on my master's thesis we tried to come up with a way to automatically record a dataset but, as @miguelriemoliveira said, it wasn't general. Yours seems like an elegant solution!

manuelgitgomes, 2023-11-17:
Hello!

Added a first version of this in branch `issue687
For now, it only looks at the changes in TFs, not yet at pattern changes.
I currently do not need those and so I will work on other stuff with higher priority,

This implementation added a new entry in the Interactive Menu of collect data to toggle the automatic data collector:
This allows the user to have the option to always turn on and off this functionality (be it to adjust the labelers, etc).
I have tested this in a `softbot2` bag file (will later test on a pr2 one), and I saw visually that it took collections at the same time I would have manually take them.
At the end, a new dataset with 55 collections was generated.
The one I generated by hand had 54, so I think this has worked great.

For this, I added a new atom_msg.srv: `ToggleAutomaticCollector and adapted the collect_data template.

You can see the implementation in this video:

https://github.com/lardemua/atom/assets/58152228/e8507af0-f1e8-45c4-9131-2d4c974009c4

manuelgitgomes, 2023-11-22:
Merged this in #705. 
This needs to add support for changing pattern points, but a new issue for that is created (#704).
I will close this issue

closed 2023-11-22 by manuelgitgomes