lardemua/atom #691: Simulate motor encoder angular offset

miguelriemoliveira, 2023-11-18:
In the sequence of #689, we decided not to include the joint states in the atom datasets, and continue to use the transformations exclusively.

We do use joint states. The joint states are in the bagfiles, and they are used in ATOM (when we are not using --use-tfs ).

These messages are read by the robot state publisher to generate the transformations, so if we want to simulate an angular error in the measurement of a motor encoder we could develop a script to add noise to these messages before they reach the robot state publisher.

From this it should be easy:

https://github.com/miguelriemoliveira/rosbag_tools

What do you think @manuelgitgomes ?

manuelgitgomes, 2023-11-20:
Yes, this might be the best way to do this without storing the joint states in atom datasets.

miguelriemoliveira, 2023-11-21:
This is operational.

Created a new bag file with noise on the elbow joint using:

    rosrun atom_calibration add_noise_to_joint_state_in_bag -bfi train.bag -bfo train_with_noise.bag -jn elbow_joint -jd 0.3

so the new bag file should have the joint_states, joint "elbow", a 0.3 radians of error. 

When running playbag, the /tf topic (produced by the robot state publisher from the incorrect joints states) should now have a value which is different from the tf_ground_truth.

The following print shows this:

[image]

commit 84a6da41 (2023-11-22): Close #691. Script to change joint state
closed 2023-11-22 by miguelriemoliveira