lardemua/atom #692: Explicitly save the ground truth on the atom datasets

miguelriemoliveira, 2023-11-18:
Right now we have the ground truth when using simulated systems, but it is stored in a strange way.

We have first guess transformations, **which we know are the ground truth** when we used a simulation to produce the bag file, and we also know that we did not change anything either using the set initial position, the --nig flag during calibration, or what will eventually come from #691.

miguelriemoliveira, 2023-11-19:
Working on branch https://github.com/lardemua/atom/tree/miguelriemoliveira/issue692

manuelgitgomes, 2023-11-20:
Yes, this change is a long time coming, as I remember having discussions about this in the past.

One thing to take into account is either to document this so that the user knows this is an option, or to add a parameters to the config.yml for simulation or gt transformations.

miguelriemoliveira, 2023-11-20:
Hi @manuelgitgomes ,

I am already half way through the implementation. I am using a flag in the configure script.

miguelriemoliveira, 2023-11-20:
Implemented. We now can get atom datasets with ground truth data.

[image]

PR #701 (2023-11-20): Miguelriemoliveira/issue692
Fixes #692
closed 2023-11-20 by miguelriemoliveira