lardemua/atom #698: Confusing method in configure_calibration_pkg

miguelriemoliveira, 2023-11-19:
https://github.com/lardemua/atom/blob/98a15dc121c76d1f3ded247d58587bd007f9bb47/atom_calibration/scripts/configure_calibration_pkg#L38

manuelgitgomes, 2023-11-22:
This is done because the files for the lidar3D evaluation use lidar.

I can move this along to other function.

closed 2023-11-22 by manuelgitgomes

miguelriemoliveira, 2023-11-22:
Hi @manuelgitgomes ,

Thanks to tackling this. I did not understand your solution tough. What do you mean by:

> I can move this along to other function

manuelgitgomes, 2023-11-22:
> I did not understand your solution tough. What do you mean by:

Basically, I was saying that the `lidar3d` modality would be only `lidar` because the evaluation files are named `lidar_to_rgb` and not `lidar3d_to_rgb`.
So, this was coded as something like `f'{source_modality}_to_{target_modality}'`

To solve this, I created this dictionary:
https://github.com/lardemua/atom/blob/30aba658f3a8c9f184e2a7fb2987b2ffeb5df528/atom_calibration/scripts/configure_calibration_pkg#L609-L620

Where you can check with your combination of modalities what would be the correct file to use.
Is this more clear now?

miguelriemoliveira, 2023-11-22:
Yes. Thanks.