lardemua/atom #704: Automatic data collection option for moving pattern

manuelgitgomes, 2023-11-22:
As implemented in #687, automatic data collection only works for moving robots that have tfs changing over time.
In the case of still systems, such as larcc (without the robot), this does not work it.
It needs a method that tests the pattern detection and if they are changing.
If they aren't, collect a collection.

miguelriemoliveira, 2023-11-22:
Thanks @manuelgitgomes, 

My suggestion is to create an issue for that, even if you will not address it right now.

manuelgitgomes, 2023-11-22:
> My suggestion is to create an issue for that, even if you will not address it right now.

This is that issue!

miguelriemoliveira, 2023-11-22:
> This is that issue!

oooops. Sorry, reading through 100 emails when I got home. Since you are guilty of about 50% of those I will not apologize :-)