lardemua/atom #708: Create a dataset management tool

manuelgitgomes, 2023-11-23:
As stated in #687, it would be good to have a dataset management tool.
Possible functionalities are:

- [ ] Dataset inspections (already done by `inspect_atom_dataset`)
- [ ] Dataset renaming (already done by `rename_atom_dataset`)
- [ ] Collection deletion
- [ ] Dataset split (into two or more datasets)
- [ ] ...

Feel free to add more if you remember anything more.

I am thinking about doing a simple CLI interface for this, where the user runs a command and selects the dataset, something like:

`rosrun atom_calibration dataset_management --dataset /path/to/dataset`

And in here a menu appears, something like

```
Welcome to the dataset manager.
Please choose the operation you want to perform by clicking on the respective letters:

(I) Dataset Inspection
(R) Dataset Renaming
(D) Collections Deletion
(S) Dataset Split

```

What do you think @miguelriemoliveira and @Kazadhum?

miguelriemoliveira, 2023-11-23:
Hi @manuelgitgomes ,

I like the idea, but am not sure about the following: right now we have one separate script for every functionality.
This is the best for maintenance and scalability.

The idea of putting together all these functionalities is nice, but if we are going to do it should we consider an application with some frontend? 

I am not sure which is best, but in any case I would assign low priority to this...

Kazadhum, 2023-11-23:
Hi @manuelgitgomes ! This sounds like a great idea! @miguelriemoliveira 's suggestion seems like a good decision too, for increased modularity.

manuelgitgomes, 2023-11-24:
Hello!

> This is the best for maintenance and scalability.

That is a good, point, haven't thought about that.
I agree with you.
Should we close this?

miguelriemoliveira, 2023-11-24:
Either close or leave open assigning low priority.

miguelriemoliveira, 2023-11-28:
I will close this for now. We have so many other things to work on.

closed 2023-11-28 by miguelriemoliveira