lardemua/atom #709: Add tams small chessboard dae to atom

miguelriemoliveira, 2023-11-25:
To make some experiments related with #706 I need this. 

@manuelgitgomes can you please add it to atom?

miguelriemoliveira, 2023-11-25:
When running 

    rosrun tams_pr2_atom_calibration configure

I get

```terminal
_______________________________________________________

           █████╗ ████████╗ ██████╗ ███╗   ███╗ 
          ██╔══██╗╚══██╔══╝██╔═══██╗████╗ ████║ 
          ███████║   ██║   ██║   ██║██╔████╔██║ 
          ██╔══██║   ██║   ██║   ██║██║╚██╔╝██║ 
   __     ██║  ██║   ██║   ╚██████╔╝██║ ╚═╝ ██║    _  
 / _|    ╚═╝  ╚═╝   ╚═╝    ╚═════╝ ╚═╝     ╚═╝   | |    
 | |_ _ __ __ _ _ __ ___   _____      _____  _ __| | __ 
 |  _| '__/ _` | '_ ` _ \ / _ \ \ /\ / / _ \| '__| |/ / 
 | | | | | (_| | | | | | |  __/\ V  V / (_) | |  |   <  
 |_| |_|  \__,_|_| |_| |_|\___| \_/\_/ \___/|_|  |_|\_\ 
 https://github.com/lardemua/atom

Configuring calibration package tams_pr2_atom_calibration
_______________________________________________________

Loading config_file /home/mike/workspaces/catkin_ws/src/calibration/robots/tams_pr2/tams_pr2_atom_calibration/calibration/config.yml

Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 108, in <module>
    config = atom_core.config_io.loadConfig(args['config_file'])
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/config_io.py", line 151, in loadConfig
    fullpath, name, uri = uriReader(config['calibration_pattern']['mesh_file'])
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/config_io.py", line 93, in uriReader
    assert (os.path.exists(fullpath)), Fore.RED + fullpath + ' does not exist.'
AssertionError: /home/mike/workspaces/catkin_ws/src/calibration/atom/atom_worlds/pattern/models/chessboard_200x214_5x4_25/chessboard_200x214_5x4_25.dae does not exist.
```

manuelgitgomes, 2023-11-27:
Hello!

Yes, I have but I need to make some changes to the pattern configuration.

Also, can you briefly explain to me how you apply a texture to a mesh?
I tried doing it and could not achieve it, if you could help me, that would be great.

miguelriemoliveira, 2023-11-27:
zoom?

manuelgitgomes, 2023-11-27:
@miguelriemoliveira yes!

manuelgitgomes, 2023-11-27:
Hello @miguelriemoliveira 

I have done it, however the orientation is still incorrect. I will try to correct it!

manuelgitgomes, 2023-11-27:
@miguelriemoliveira  well, I tried changing this is blender and it did not seem to change in rviz. Did you change it in meshlab?

miguelriemoliveira, 2023-11-27:
Orientation what do you mean? Where the reference system is? That I know how ti do in meshlab.

I have family at home now, so only tonight I can get back to this.

miguelriemoliveira, 2023-11-28:
Can you push the pattern with texture. I can then use meshlab to correct its origin ...

manuelgitgomes, 2023-11-28:
> Can you push the pattern with texture. I can then use meshlab to correct its origin ...

It is already pushed!

miguelriemoliveira, 2023-11-28:
To run I need #720  fixed.

I need to see the detections of the pattern in the images to figure out where the reference system should be.

[image]

closed 2023-11-28 by miguelriemoliveira

miguelriemoliveira, 2023-11-28:
I was able to run the tams_pr2_atom_calibration playbag.launch and test the labeler. It is confirmed that the reference frame is the one I thought before.

[image]