lardemua/atom #711: Calibration of dynamic transformations

miguelriemoliveira, 2023-11-25:
Required by #703 .

When calibrating the system in #703, we wanted to calibrate the dynamic joints from the manipulator as well.

First problem is a safeguard in configure. When running

    rosrun rgb_hand_eye_system_calibration configure -cgt

we get:

```terminal
Error: additional_tf  base_link_inertia_to_shoulder_link defines transformation base_link_inertia to shoulder_link to be calibrated but this transformation is not static and therefore cannot be calibrated.
```

This protection makes sense because optimizing dynamic transformations would mean to have a different transformation for each collection. That does not make sense, as the optimization would have to be cast with one different transformation per collection (as in the case of the pattern).

miguelriemoliveira, 2023-11-25:
Assuming this is a revolute joint, the solution I think is to try to optimize the revolute angle bias that produces all these transfomations ...

miguelriemoliveira, 2023-11-25:
Added the joints to the datasets. Here's a snapshot:

[image]

miguelriemoliveira, 2023-11-26:
Finally I was able to find a way to compute the transformation after the joint configuration and the joint position.

Here's an image where the longer, thinner reference frame is computed by our new method (to be included in the calibration) and the smaller, think frame is provided by robot state publisher.

[image]

miguelriemoliveira, 2023-11-27:
Good news. I have created a dataset where the elbow joint is affected by -0.1 radians. Here is the [train_with_noise_elbow_joint](https://drive.google.com/file/d/1b53p-zRzkAnH1Ty12tZw9QvrFocoEWKD/view?usp=sharing).

Then I configured ATOM to run a joint calibration, in this case only specifying, for the elbow joint, the position_bias joint parameter to be optimized.

Results without the joint calibration are:

[image]

And with the joint calibration enabled we get:

[image]

And what is even better, the parameter values are:

```terminal
Parameters:
                                               Group        x0         x      data  Min  Max
flange-rgb_hand_link_x          flange-rgb_hand_link  0.058680 -0.020206 -0.020206 -inf  inf
flange-rgb_hand_link_y          flange-rgb_hand_link -0.043391 -0.000448 -0.000448 -inf  inf
flange-rgb_hand_link_z          flange-rgb_hand_link  0.021106  0.064461  0.064461 -inf  inf
flange-rgb_hand_link_r1         flange-rgb_hand_link  0.104833 -0.000016 -0.000016 -inf  inf
flange-rgb_hand_link_r2         flange-rgb_hand_link  0.094841  0.000728  0.000728 -inf  inf
flange-rgb_hand_link_r3         flange-rgb_hand_link -0.104833 -0.000876 -0.000876 -inf  inf
world-pattern_link_x              world-pattern_link -0.782967 -0.849128 -0.849128 -inf  inf
world-pattern_link_y              world-pattern_link  0.251381  0.291022  0.291022 -inf  inf
world-pattern_link_z              world-pattern_link  0.820644  0.808925  0.808925 -inf  inf
world-pattern_link_r1             world-pattern_link  3.064886  3.121485  3.121485 -inf  inf
world-pattern_link_r2             world-pattern_link  0.033466  0.035512  0.035512 -inf  inf
world-pattern_link_r3             world-pattern_link  0.120324 -0.009249 -0.009249 -inf  inf
elbow_joint-position_bias  elbow_joint-position_bias  0.000000  0.100286  0.100286 -inf  inf
```

which shows value 0.100286 for the elbow_joint-position_bias. Note that we affected the joint by -0.1, and the optimization was able to estimate a bias of 0.1 to compensate for that.

These are very good news. The first joint optimization working on atom!

miguelriemoliveira, 2023-11-27:
Another experiment, now with a dataset that contains angular bias changes in all joints. Its [train_with_noise_all_joints](https://drive.google.com/file/d/1LJEod3ET8PwRbehR6d5j0rjOeLf95kyg/view?usp=sharing).

Calibrating without joint calibration include gives:

[image]

and running the calibration of all joints turn out in:

[image]

The estimated parameters are:

```terminal
Parameters:
                                                                           Group        x0         x      data  Min  Max
flange-rgb_hand_link_x                                      flange-rgb_hand_link  0.053066 -0.020155 -0.020155 -inf  inf
flange-rgb_hand_link_y                                      flange-rgb_hand_link  0.002721  0.000903  0.000903 -inf  inf
flange-rgb_hand_link_z                                      flange-rgb_hand_link -0.003219  0.064389  0.064389 -inf  inf
flange-rgb_hand_link_r1                                     flange-rgb_hand_link -0.104833 -0.024547 -0.024547 -inf  inf
flange-rgb_hand_link_r2                                     flange-rgb_hand_link  0.094841  0.000618  0.000618 -inf  inf
flange-rgb_hand_link_r3                                     flange-rgb_hand_link  0.104833 -0.000546 -0.000546 -inf  inf
world-pattern_link_x                                          world-pattern_link -0.874833 -0.850234 -0.850234 -inf  inf
world-pattern_link_y                                          world-pattern_link  0.300491  0.292423  0.292423 -inf  inf
world-pattern_link_z                                          world-pattern_link  0.825617  0.809207  0.809207 -inf  inf
world-pattern_link_r1                                         world-pattern_link -3.127298 -3.161783  3.120820 -inf  inf
world-pattern_link_r2                                         world-pattern_link -0.064991 -0.042369  0.041820 -inf  inf
world-pattern_link_r3                                         world-pattern_link  0.057713  0.007643 -0.007544 -inf  inf
joint-elbow_joint-position_bias                  joint-elbow_joint-position_bias  0.000000  0.030242  0.030242 -inf  inf
joint-shoulder_lift_joint-position_bias  joint-shoulder_lift_joint-position_bias  0.000000 -0.035366 -0.035366 -inf  inf
joint-shoulder_pan_joint-position_bias    joint-shoulder_pan_joint-position_bias  0.000000  0.013998  0.013998 -inf  inf
joint-wrist_1_joint-position_bias              joint-wrist_1_joint-position_bias  0.000000 -0.049880 -0.049880 -inf  inf
joint-wrist_2_joint-position_bias              joint-wrist_2_joint-position_bias  0.000000 -0.010221 -0.010221 -inf  inf
joint-wrist_3_joint-position_bias              joint-wrist_3_joint-position_bias  0.000000  0.054521  0.054521 -inf  inf
```

which fits the perturbation we added in the joint angles, with this command:

    rosrun atom_calibration add_noise_to_joint_state_in_bag -bfi train.bag -bfo train_with_noise_all_joints.bag -jn shoulder_lift_joint elbow_joint wrist_1_joint wrist_2_joint wrist_3_joint shoulder_pan_joint -jb 0.034 -0.03 0.05 0.01 -0.03 -0.01

closed 2023-11-27 by miguelriemoliveira

manuelgitgomes, 2023-11-27:
Very nice work @miguelriemoliveira, congratulations!

It seems like atom beats a new challenge once again ;)

miguelriemoliveira, 2023-11-27:
Thanks. Now I am curious to see the impact on the  pr2 calibration!