lardemua/atom #712: Verify if joints to be calibrated do not collide with transformations to be calibrated

miguelriemoliveira, 2023-11-25:
In configure_calibration_pkg

manuelgitgomes, 2023-11-27:
Shouldn't this be covered by stating that we cannot calibrate dynamics transformations?

Or is this a safe way to restrict joint calibration for revolute/prismatic joints?

miguelriemoliveira, 2023-11-27:
Hi @manuelgitgomes ,

you are right, possibly the calibration of transformations and joints will not collide.

However, there are several nuances, and also we don't know how this is going to evolve, so I am of the opinion that it is better to explicitly test if we are simultaneously calibrating a transformation and a corresponding joint, and throw an error if so.

Note for example that we  calibrate the world to pattern transformation, which is not static. 
Also, for now we can only calibrate revolute joints, but we could in theory calibrate static joints.

miguelriemoliveira, 2023-11-28:
Implemented.

commit 6978acee (2023-11-28): Close #712 Verify if joints to be calibrated do not collide with transformations to be calibrated
closed 2023-11-28 by miguelriemoliveira