lardemua/atom #717: How to write joint calibration to urdf

miguelriemoliveira, 2023-11-27:
Having atom calibrating joints, now we must figure out how to pass the results of calibration to the xacro.

The joint has a calibration field:

From http://wiki.ros.org/urdf/XML/joint

we have:

> <calibration> (optional)
> 
> The reference positions of the joint, used to calibrate the absolute position of the joint.
> rising (optional)
> 
> When the joint moves in a positive direction, this reference position will trigger a rising edge.
> falling (optional)
> 
> When the joint moves in a positive direction, this reference position will trigger a falling edge.

I don't really get why we need a rising and a falling. Why not have a positive / negative bias ?

@manuelgitgomes can you understand this?

miguelriemoliveira, 2023-12-03:
Hi @manuelgitgomes ,

You told the calibration only uses the rising. But will the transformations be compensated by this bias when published by the robot state publisher? Can you run a small test to figure out if this occurs?

manuelgitgomes, 2023-12-04:
> But will the transformations be compensated by this bias when published by the robot state publisher?

Unfortunately not.
These calibration tags only affect the joint state controller, that receives information from the robots drives and publishes the joint states.
Therefore, it seems that there is not straight forward way of playing a rosbag that reads the urdf and places everything in its correct place.
The only way I am seeing now is to use the script to add noise to joint states to have new joint states on the bagfiles and verify everything.

miguelriemoliveira, 2023-12-04:
> The only way I am seeing now is to use the script to add noise to joint states to have new joint states on the bagfiles and verify everything.

We could also change the coordinate system of the axis to rotate the value of the bias, and fill the rising with nothing, so the joints keep being published with an error but the xacro compensates for that.

What do you think?

miguelriemoliveira, 2023-12-07:
Hi @manuelgitgomes ,

Thanks. You will use that for running the pr2 live?

miguelriemoliveira, 2023-12-30:
I think the solution we agreed upon in to have a general urdf exporter, a general yaml exporter, and a tams exporter.

closed 2023-12-30 by miguelriemoliveira