lardemua/atom #719: Create JointStates buffer

manuelgitgomes, 2023-11-28:
Different JointStates messages with different joints can be published to the same topic.

This is the case in the PR2, as one message is for the Shadow Hand and the other is for the rest of the robot.

As the collection code is, we only look at the last JointStates message.
This can cause errors, p.e., if you were looking for the head joint in the Shadow Hand message.

To try and solve this, I will create a buffer that records the last x messages (maybe 5 or 10).
From this buffer, we will search for the joint state from the most recent to the eldest.

miguelriemoliveira, 2023-11-28:
Makes sense... let me know if you need help.

manuelgitgomes, 2023-11-28:
Done!

Created a cache that, at the moment, store the 10 last joint states received.
From here, it check if the specific joint is in the most recent message.
If not, it checks the second most recent, and so on and so forth.

miguelriemoliveira, 2023-11-28:
Hi @manuelgitgomes , 

thinking about it, I think I may have a better idea. Store a dictionary where keys are joint names, and values are joint positions. Every time a joint state msg is received, iterate joint names , update that dictionary. This will only update joints that are present in the joint_state message.

Then, in the save collection method, use the dictionary.

manuelgitgomes, 2023-11-28:
> Then, in the save collection method, use the dictionary.

It is much more organized yes, I can reopen this and do it.