lardemua/atom #722: Create joint_selection_function and additional_tf_selection_function

manuelgitgomes, 2023-11-28:
The objective in here is to achieve a similar functionality than `collection_selection_function` and  `sensor_selection_function`.

This would allow us to get more control over the calibration, without recording new datasets.
It would also allow us to create ablation studies more easily.
In the case of #706, I want to remove the head joints to verify what difference they make on the calibration.

manuelgitgomes, 2023-11-28:
Created selection function for joints and additional_tfs

Exactly the same way as it is done for sensors and collections!

manuelgitgomes, 2023-11-28:
Sorry @miguelriemoliveira, I just now noticed that you had an argument to skip all joint calibration.

That would be deprecated now with the joint selection function, right?

miguelriemoliveira, 2023-11-28:
> That would be deprecated now with the joint selection function, right?

Right. Better remove.