lardemua/atom #727: Solving sensor selection function after pattern definition

manuelgitgomes, 2023-11-29:
The idea in this issue is to take care of this dependency:

https://github.com/lardemua/atom/blob/4ac1fe345e47b0a3e07468ae0792e4ff4002cfa0/atom_calibration/scripts/calibrate#L140-L145

This is bothering me at the moment because, on #706, the initial estimate of the pattern is given by a sensor that is not been used for calibration, and that sensor is not calibrated.
This results in a very incorrect initial guess of the pattern that might lead to the optimization falling on local minima.
This does not happen on #706, because I tested it using a first guess from the correct sensor.

I think the better way to do this is to filter the sensors first and only then to define the pattern.
If that was not possible, try again using the original dataset.
What do you think @miguelriemoliveira?
Do you have any better idea?

miguelriemoliveira, 2023-11-29:
Hi @manuelgitgomes ,

the problem was how to create a first guess for the pattern pose when the  current calibration setup is not using an rgb camera. In those cases we  need an rgb sensor to compute the patterns pose, and only after we can remove the rgb sensor from the dataset.

Your solutions seems to work perfectly. First filter and try to find an rgb sensor in the filtered dataset. 
If you find it, use it for calibration. If not, search for an rgb sensor in the original (complete) dataset.
Seems like a sound logic.

The ideal way of solving this would be to create a mechanisms to create a pattern pose first guess also from other modalities. Or to have an interactive way in which the use could say the pattern is more or less here. 
But lets leave that for now.

manuelgitgomes, 2023-11-29:
Ok @miguelriemoliveira.

I will do this small fix, but larger problem will still remain.

manuelgitgomes, 2023-11-29:
Solved by creating a condition to verify if the filtered dataset has an rgb sensor.
If the filtered dataset has an rgb sensor, create the labels from it.
If not, create the labels from a previous form of the dataset.

closed 2023-11-29 by manuelgitgomes