lardemua/atom #729: Rotation of chessboard in ATOM

manuelgitgomes, 2023-11-29:
After analyzing the OPENCV code, I verified that the some patterns are indeed independent of orientation.
This is because of the discussion we previously had about having the left corner squares black and the right corner squares white.
This is the case of the pattern here at TAMS.

OpenCV flips the detection in cases where the pattern is symmetrical, as can be seen in:

https://github.com/opencv/opencv/blob/5df28f1eaa4e7433cfb2d51898384bbb30da8468/modules/calib3d/src/calibinit.cpp#L642-L659

This is not our case.

Therefore, I think this block of code is not needed:
https://github.com/lardemua/atom/blob/30aba658f3a8c9f184e2a7fb2987b2ffeb5df528/atom_calibration/src/atom_calibration/collect/patterns.py#L25-L29

miguelriemoliveira, 2023-11-29:
Please comment it. I need to test the rgb_rgb_system_chessboard with the new change.

manuelgitgomes, 2023-11-29:
Done!

closed 2023-11-29 by manuelgitgomes