lardemua/atom #731: New stuff to support in config.yml and dataset v 2.0

miguelriemoliveira, 2023-11-29:
- [ ] Multiple calibration patterns
- [ ] Joints
- [ ] Remove joints: "" and move to format joints: 
- [ ] Remove addtional_tfs: "" and move to format addtional_tfs: 
- [ ] joints will loose parent and child and get same name as urdf
- [ ] All fields in config.yml are mandatory, even if they are not on the first level
- [ ] Add version to config

manuelgitgomes, 2023-11-29:
Additional tfs and joints are now mandatory!

miguelriemoliveira, 2023-11-29:
In #728 created a script to update config,yml files to version 2.0.

Already advanced all config files in atom examples to v2.0.

miguelriemoliveira, 2023-11-29:
Hi @manuelgitgomes , since you are working on the datasets then I should be hands off, right?

I think it is better to have a single person working on all this at the same time, as there are many interconnections ...

manuelgitgomes, 2023-11-29:
Hello @miguelriemoliveira, I will be leaving for today, so please continue.

In the last commit I created a first version of dataset conversion
This version only consists of adding additional_tfs, joints, package_name and changing the dataset version.

miguelriemoliveira, 2023-11-30:
configure_calibration_pkg seems ok. Moving on to collect_data

miguelriemoliveira, 2023-11-30:
I am halfway through correcting the collect and label script.

Only tomorrow afternoon I will come back to this.

@manuelgitgomes if you want, advance with the dataset converter. I will call when I start working again so we can sync.

manuelgitgomes, 2023-11-30:
Hello @miguelriemoliveira.

I am now going for lunch, so I will just update you on what I am doing.

I found we have some mistakes on config conversion:

- We do not add modality nor throttle to sensors that do not have them
- We convert all non string values to string (booleans for defining pattern as fixed, numbers for pattern size, etc), this could cause errors along the line. If you think not, tell me please.

I am trying to fix those

miguelriemoliveira, 2023-11-30:
Thanks. I will continue to work on the data_collector producing a correct dataset format.

Will keep you posted.

manuelgitgomes, 2023-11-30:
> Thanks. I will continue to work on the data_collector producing a correct dataset format.

I was working on that, do you want me to commit my findings and you keep working?

manuelgitgomes, 2023-11-30:
We can also change anchored_sensor from "" to null, right?

miguelriemoliveira, 2023-11-30:
Right. I think makes nor sense.

I like that in configure_calibartion_pkg we have a standarized way of testing, e.g.:

if config['joints'] is no None:
   # do stuff

miguelriemoliveira, 2023-11-30:
Hi @manuelgitgomes ,

when you leave work this afternoon please push and  let me know where you are.
I will pick it up tonight.

manuelgitgomes, 2023-11-30:
Hello @miguelriemoliveira.

Sorry, I had a meeting that took a big part of the afternoon so I had to drop this in the middle.
As I did not want to push something half-baked, I finished everything now, so only now I have pushed a lot of commits.

Trying to make a summary:

- I changed anchored sensor from '' to null and adapted everything for it
- Solved a bug that when 'toggle' was null, the toggle nodes were still launched, causing in error
- In "non-essential" scripts, I implemented the simple solution of changing the 'calibration_pattern' to the first element of 'calibration_patterns'
- I added support for multiple patterns in data collector. Should theoretically work right out of the box if we implemented now the multiple patterns. (I would still be very VERY surprised nonetheless if it did). 
To do this, I moved all the labels (`dataset['collections'][collection_key]['labels']`) to a dictionary with the name `pattern_1` (`dataset['collections'][collection_key]['labels']['pattern_1']`
- Updated the dataset updated, to be up to date with the previous changed

Please test data collection and dataset conversion.
If everything is ok, we can would now on the calibrate script.

manuelgitgomes, 2023-12-01:
Hello @miguelriemoliveira!

I will be seeing if setting initial estimate and dataset_playback are functional and if not, fixing them.

After I have done that, I will report back.

miguelriemoliveira, 2023-12-01:
Thanks. I am creating a new atom_example called eye_in_world_system so I can test everything from scratch, i.e. without the updaters.

One thing I changed to be consistent with the global philosophy is this:

```yml
  # throttle:
  #   Set throttle: desired_frame_rate. If you don't use throttle, do not enter a value, i.e. "throttle: "
  #
  # If you are using an image compressed topic such as:
  #   /world_camera/rgb/image_raw/compressed
  # you should not add the "compressed" part, use only:
  #   /world_camera/rgb/image_raw
  #

  hand_camera:
    link: "hand_camera_optical_frame"
    parent_link: "end_effector"
    child_link: "hand_camera"
    topic_name: "/hand_camera/image_raw"
    modality: "rgb"
    throttle: 10

  # You can have more than one sensor:
  another_camera:
    link: "another_camera_optical_frame"
    parent_link: "end_effector"
    child_link: "another_camera"
    topic_name: "/another_camera/image_raw"
    modality: "rgb"
    throttle:
```

I have fixed a lot of stuff already so I will push it right now.

miguelriemoliveira, 2023-12-01:
> I am working on the dataset_play, which is not functional, so please do not try to fix it :)

ok

manuelgitgomes, 2023-12-01:
Added support for multiple pattern in dataset conversion and rgb to rgb evaluation.
These are still very "primitive", only choosing the first pattern, but I think for these files it is ok.
For dataset conversions, those original datasets only support one pattern, so choosing only the first one make sense.
If needed, we can then create a flag to choose the correct pattern and use that value.
This is similar for rgb_to_rgb_evaluation, probably will be based on a single pattern for ever, so we might add a flag to choose the correct pattern and add it to the full_evaluation.launch if it comes to that.

miguelriemoliveira, 2023-12-01:
Hi @manuelgitgomes ,

thanks. I thing for the old systems, such as the rgb_rbg_system, the conversion will create **calibration_patterns** dictionary which will have a single pattern, e.g., pattern_1.

In the calibrate script (and everywhere else as far as I can see) we will always use all patterns (not for now, when we make the enhancements). Then, for the rgb_rgb_system single we have a single key in the dictionary we will calibrate using only that pattern.

Does it make sense? I am not sure why we need the flag you mention ...

We could have a flag like **csf**, like a **pattern_selection_function**, but everything should work without it, right?

manuelgitgomes, 2023-12-01:
> the conversion will create **calibration_patterns** dictionary which will have a single pattern, e.g., pattern_1.

Exactly.

> Then, for the rgb_rgb_system single we have a single key in the dictionary we will calibrate using only that pattern.

Yes, it makes sense.

> Does it make sense? I am not sure why we need the flag you mention ...

I am talking about dataset conversion, not update.
This is converting ATOM datasets to other frameworks and vice versa.
Is it more clear now?

> We could have a flag like **csf**, like a **pattern_selection_function**, but everything should work without it, right?

This is a good idea for the calibration!

miguelriemoliveira, 2023-12-01:
> I am talking about dataset conversion, not update.
> This is converting ATOM datasets to other frameworks and vice versa.
> Is it more clear now?

Ah, now I get it. But I think it does not apply.

The dataset update script is supposed to move from version 2.0 to 3.0. 
In version 2.0 you can only have one pattern (define in the **calibration_pattern**), which during conversion should be placed in the **calibration_patterns** under a hardcoded key such as **pattern_1**.

What do you think?

manuelgitgomes, 2023-12-01:
> In version 2.0 you can only have one pattern (define in the **calibration_pattern**), which during conversion should be placed in the **calibration_patterns** under a hardcoded key such as **pattern_1**.

This is what is already being done on the dataset update (at least that is what I intended that it should do).
Is it not?

miguelriemoliveira, 2023-12-01:
> This is what is already being done on the dataset update (at least that is what I intended that it should do).
> Is it not?

Yes, I assume this is what you did and it is the correct way. What I do not understand is why we need this flag

> If needed, we can then create a flag to choose the correct pattern and use that value.

We will always have a single calibration pattern, always to be mapped to one key in the dictionary called pattern_1. 
Perhaps I am missing something ...

manuelgitgomes, 2023-12-01:
> We will always have a single calibration pattern, always to be mapped to one key in the dictionary called pattern_1. Perhaps I am missing something ...

I am talking about converting ATOM dataset to some other dataset.
Imagine the time ATOM really has various patterns, and other datasets only support one.
Which one of ATOM should be chosen?
This is why I proposed to add a flag for that, but only in the future when proven needed.

manuelgitgomes, 2023-12-01:
@miguelriemoliveira I think calibration visualization is very similar to dataset_playback visualization, right?

If so, please do not work on that as I am currently working on the visualization of dataset_playback, so after then I can reuse a lot of code for that.

miguelriemoliveira, 2023-12-01:
I @manuelgitgomes ,

I remember there were some distinctions.For example, during dataset playback you want to show a single collection, whereas in the calibrate you want to see all collections at once. I think that was the main reason why I created a visualization for each.

In any case, if we have common features, we can reuse the code, yes.

miguelriemoliveira, 2023-12-01:
right, be careful because I remember splitting into two for several reasons.

manuelgitgomes, 2023-12-01:
@miguelriemoliveira, what do you think we should use to visually distinguish different pattern detections?
Use different colormaps?

miguelriemoliveira, 2023-12-01:
No problem. I will.

manuelgitgomes, 2023-12-01:
With the previous commit, I introduced a bug that did not publish labeled messages for all sensors except the first.
The same would happen when saving data, only saving for the first sensor.
I also solved a previous bug for additional_data.

miguelriemoliveira, 2023-12-01:
Just pushed a commit with lots of changes to support multiple patterns in calibrate.
Will later create an atom example with multiple patterns to test, but for now the eye_in_world seems (#739) operational.

miguelriemoliveira, 2023-12-01:
In any case, I will collect a new dataset and try to calibrate again.

miguelriemoliveira, 2023-12-01:
> In any case, I will collect a new dataset and try to calibrate again.

Collected a new dataset, then calibrated. Everything works fine. Subpixel error.

manuelgitgomes, 2023-12-01:
Hello @miguelriemoliveira, have you changed anything in calibration?

To end dataset_playback I need to change a function that affect calibration as well: `getPointsInDepthSensorAsNPArray`.
Just to check if you have already changed it.
Because I have two options right now:

- I change it and hardcode `pattern_1` when it is called on calibration
- I change the whole calibration script, which does not seem a good option for today, as it might take a while.

miguelriemoliveira, 2023-12-01:
I @manuelgitgomes ,

I was working in calibration and changed a lot of stuff there to make it work with multiple patterns. I always used loops to iterate all the patterns, never the "use the first in the list option".

That funtion in particular I think I did not change because I was not using the depth modality I didn't notice it. 

But I will change it in accordance to the others (I know how to).

The calibration script is almost all changed to operate with multiple patterns. Of course I am expecting to have multiple errors in the near future but we are moving ahead!

miguelriemoliveira, 2023-12-01:
I will change the function and get back to you.

miguelriemoliveira, 2023-12-01:
Done. This also works with multiple patterns.

manuelgitgomes, 2023-12-01:
I think that finally dataset_playback is done (for now...)

As I said, I fully reconfigured the `graphics` dictionary.
It now has the keys:

- colormap - it defines the colormaps for each pattern for rgb images and for each collection
- sensors - it defines the publishers and topics to publish for each sensor
- ros - same as before
- labels - has the labels organized by pattern_key

I have some things that, unfortunately, I could not change now, as I believe we needed to be more developed in the multi-pattern approach.

Some things are still to be done and are marked as TODO on comments:
- Depth and lidar manual labeling only works for a single pattern, as we would need an external input to define which pattern we were labeling
- The infrastructure for defining different colormaps for each pattern is there, but it is not yet implemented

Nevertheless, it works for PR2 and softbot2.

miguelriemoliveira, 2023-12-01:
> I have some things that, unfortunately, I could not change now, as I believe we needed to be more developed in the multi-pattern approach.

Great. That's a good first version. For the multipattern to be tested properly, we need to create an atom_example. I will do so in the near future.

miguelriemoliveira, 2023-12-23:
Multiple calibration patterns are now supported. Check #781. Closing this issue as all checkboxes are ticked.

closed 2023-12-23 by miguelriemoliveira