lardemua/atom #734: not sure how to solve this one well ... if at least one is fixed?

miguelriemoliveira, 2023-11-30:
```
        # TODO not sure how to solve this one well ... if at least one is fixed?
```

https://github.com/lardemua/atom/blob/234402379ca3f290aa3efb6a43dbef80b25afa17/atom_calibration/scripts/configure_calibration_pkg#L765

manuelgitgomes, 2023-11-30:
Well, supposing evaluation would be done w.r.t. various patterns, we would have to run an evaluation script for each combination of sensors + pattern.

Therefore, I think this block of code:

https://github.com/lardemua/atom/blob/234402379ca3f290aa3efb6a43dbef80b25afa17/atom_calibration/scripts/configure_calibration_pkg#L725-L789

Should run inside a for loop, for each pattern defined.

manuelgitgomes, 2023-11-30:
I applied the suggestion I gave above and corrected the launch file for the atom_examples

miguelriemoliveira, 2023-11-30:
Great. Thanks.