lardemua/atom #735: rgb_depth_system bag file link goes to rgb_rgb_system

manuelgitgomes, 2023-11-30:
Hello @miguelriemoliveira.

The link for the bagfile for the `rgb_depth_system` goes to the bagfile of the rgb_rgb_system.
Can you fix it?

manuelgitgomes, 2023-11-30:
How come? 
You don't have the original bagfile?

I think my dataset converter already works for the changes we have applied.
There might be more changes that are not yet implemented and so it might be soon outdated.
To run it, just use:

`rosrun atom_calibration update_dataset_from_version_2_to_3 -json path/to/dataset -pn package_name`

miguelriemoliveira, 2023-11-30:
Sorry, I understood dataset and not bagfile. I will fix the link soon,

Also, I will use your script to update the datasets and update those links as well.

miguelriemoliveira, 2023-11-30:
Done.