lardemua/atom #737: Update calibrate to work with datasets v3.0

miguelriemoliveira, 2023-11-30:
To make rgb_rgb_system entire pipeline work, I need this. 

I will work on this. @manuelgitgomes , if you are working on this as well let me know.

miguelriemoliveira, 2023-11-30:
When running:

    rosrun atom_calibration calibrate -json $ATOM_DATASETS/rgb_rgb_system/rgb_rgb_system_example_train_dataset/dataset.json -v

I get

```
After filtering, will use 4 collections: ['000', '001', '002', '003']
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 607, in <module>
    main()
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 146, in main
    dataset["patterns"] = patterns.createPatternLabels(args, dataset)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/calibration/patterns_config.py", line 68, in createPatternLabels
    nx = dataset['calibration_config']['calibration_pattern']['dimension']['x']
KeyError: 'calibration_pattern'
```

manuelgitgomes, 2023-11-30:
Be careful.

From what I am changing on collect_data, labels might be different, as they most likely will be inside a dictionary, to support various pattern.
This change is not yet expected on my dataset conversion script, so tread lightly when messing with labels, as they are most likely to be changed.

miguelriemoliveira, 2023-11-30:
Hum, what do you mean? That your current dataset converter will change? I thought that, since I was using the new dataset v3.0, I could use it to test the caibration. But if not, then might as well wait for you to finish the collect data (and then the new version of the conversion to be consistent).

I will push my changes, related with the change from calibration_pattern to calibration_patterns. Once you finish your part, we can go back to this.