lardemua/atom #741: Moving the pattern pose estimation from calibrate into dataset_collect

miguelriemoliveira, 2023-12-01:
Hi @manuelgitgomes ,

just an idea I had for discussion.

You know that in the calibrate script, we run the pattern pose estimation from an rgb camera that is in the dataset.

https://github.com/lardemua/atom/blob/9fbc47643cd75302dbf9222c6c8a380135e4f2d1/atom_calibration/src/atom_calibration/calibration/patterns_config.py#L40

This creates a patterns key in the dataset, which is then used to optimize. This approach has the problem of having to have an rgb sensor in the system, otherwise we cannot estimate a first guess for the pattern pose. Moreover, it creates those intricate dependencies in the filtering of collections that you solved in #727 .

My suggestion is: Move initial estimation of pattern poses to the dataset_collector and labeler. The advantages are:

1. We can show during collection the estimated patter pose
2. We can manually correct the estimated pattern pose, either during the collect stage or in the dataset_playback
3. We would loose those intricate dependencies in the filtering of dataset
4. We can drop the requirement of having an rgb sensor to create a first guess pattern pose:
4.1  If there are no cameras we can set the pattern's pose random, and ask the user to fix it.
4.2 We could even develop a pattern pose estimation based on lidar. If it fails, the user should fix it.

What do you think about this?

miguelriemoliveira, 2023-12-01:
Your comment disappeared? I received an email but now its now here ... did you delete?

miguelriemoliveira, 2023-12-01:
> I have some questions though.
> Would we still have a fixed estimated pose or would it be different during the collection stage?
> I.e., would you capture the estimated pattern pose from a camera in the beginning and keep it constant during the collection (adding the option for the user to manually change it)?
> Or would this be changing whenever new images reach the sensors?
> If so, how would we choose which sensor to prioritize?
> Should we average the estimated poses?

I think it should be handled like the labels are. We are continuously labeling sensor data, and when the user decides to save a collection we record the most recent label (which should be the one with the closest timestamp to the timestamp of the collection).

The problem of selecting which sensor to use we already have it now, and solve it trivially by selecting the first rgb sensor that appears on the list.

> And how would this alter the dataset?
> Will we have a new parameters for each collection that is the pose of the pattern?

Right now we detect the pattern and produce this:

[image]

The idea is to produce the same information but during the collect stage. In sum, the dataset.json will be similar to the atom_calibration.json, as it will also have the key patterns.

> Would we add this to the transforms dictionary using the parent_link and pattern_link used on config?

I think we already take care of this. Just do like we did so far.

> If we have different transformations for each collection but the pattern is fixed, which one would we choose?
> Would it be random? Would we average them? Would we calculate the residuals for all and chose the one with the lowest residuals?

That's a good question. Right now, from what I saw in the code, we compute an initial_transform for each collection.
Then, if we have a fixed pattern, we use the initial_transformation of the selected_collection_key. So we use a single collection.

https://github.com/lardemua/atom/blob/9fbc47643cd75302dbf9222c6c8a380135e4f2d1/atom_calibration/scripts/calibrate#L328-L345

It is a trivial solution, but its only one to come up with a first guess, and we've seen that it has been good enough for now.

miguelriemoliveira, 2023-12-02:
This is operational. Just calibrated eye_in_world_system after a collect data which computes the pattern poses.
Here's the dataset produced by collect _data.

[image]

The problem now is that all our datasets are deprecated, since they do not contain the patterns key.
To solve that, as we discussed, the dataset update script should be able to create this information when updating.

I opened #746 for that and will work on it. I know more or less what to do, since I did it just now in collect_data...

miguelriemoliveira, 2023-12-02:
Messed up the commit message. Here's the commit,

https://github.com/lardemua/atom/commit/2e321656632f4a6fcec985ff966722e6ff90e1b6

closed 2023-12-02 by miguelriemoliveira

miguelriemoliveira, 2023-12-02:
Removing pattern pose estimation from calibrate script ...

manuelgitgomes, 2023-12-04:
Great job!