lardemua/atom #745: Error in dataset updater

miguelriemoliveira, 2023-12-01:
Hi @manuelgitgomes ,

trying to update the datasets for the rgb_rgb_system. Got this error when running 

    rosrun atom_calibration update_dataset_from_version_2_to_3 -json dataset.json -pn rgb_rgb_system_calibration

```
Loading dataset dataset.json
Backing up your dataset
output_folder is: 
[ WARN:0@0.415] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_left_000.jpg'): can't open file for writing: permission denied
Saved file /rgb_left_000.jpg.
[ WARN:0@0.423] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_right_000.jpg'): can't open file for writing: permission denied
Saved file /rgb_right_000.jpg.
[ WARN:0@0.430] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_left_001.jpg'): can't open file for writing: permission denied
Saved file /rgb_left_001.jpg.
[ WARN:0@0.438] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_right_001.jpg'): can't open file for writing: permission denied
Saved file /rgb_right_001.jpg.
[ WARN:0@0.446] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_left_002.jpg'): can't open file for writing: permission denied
Saved file /rgb_left_002.jpg.
[ WARN:0@0.453] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_right_002.jpg'): can't open file for writing: permission denied
Saved file /rgb_right_002.jpg.
[ WARN:0@0.461] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_left_003.jpg'): can't open file for writing: permission denied
Saved file /rgb_left_003.jpg.
[ WARN:0@0.469] global /io/opencv/modules/imgcodecs/src/loadsave.cpp (773) imwrite_ imwrite_('/rgb_right_003.jpg'): can't open file for writing: permission denied
Saved file /rgb_right_003.jpg.
Saved json output file to dataset.json.old.
Adding package name rgb_rgb_system_calibration to calibration_config[OK]
Adding empty joints parameter to calibration_config[OK]
Adding empty additional_tfs parameter to calibration_config[OK]
Replacing single pattern on calibration_config with a dictionary of pattern with a single key[OK]
Adding package name rgb_rgb_system_calibration to _metadata[OK]
Adding empty joint dictionary to collections[OK]
Moving labels inside a dictionary specific for the pattern used[OK]
Adding version 3.0 to _metadata[OK]
Saving your updated dataset
output_folder is: 
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/utilities/update_dataset_from_version_2_to_3", line 112, in <module>
    saveResultsJSON(args['json_file'], dataset)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 177, in saveResultsJSON
    createDataFile(dataset, collection_key, sensor, sensor_key, output_folder)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 213, in createDataFile
    cv_image = getCvImageFromDictionary(dataset['collections'][collection_key][data_type][sensor_key])
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 319, in getCvImageFromDictionary
    return bridge.imgmsg_to_cv2(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 174, in imgmsg_to_cv2
    im = np.ndarray(shape=(img_msg.height, img_msg.width, n_channels),
TypeError: buffer is too small for requested array
```

miguelriemoliveira, 2023-12-01:
This line seems strange:

 cv_image = getCvImageFromDictionary(dataset['collections'][collection_key][data_type][sensor_key])

manuelgitgomes, 2023-12-01:
> cv_image = getCvImageFromDictionary(dataset['collections'][collection_key][data_type][sensor_key])

This is correct, as data_type is 'data', which is as datasets are structured.

I think this last commit will solve this, but I had not way of recreating the error, so please test on your side.
What I did was to replace the saving method.
Instead of saving everything (images, .pcd, etc), the script is now only saving the dataset.json, which is the only thing being altered.

miguelriemoliveira, 2023-12-02:
Working fine now. Thanks.

closed 2023-12-02 by miguelriemoliveira