lardemua/atom #750: Full evaluation crashes in rgb_rgb_system

miguelriemoliveira, 2023-12-02:
When running 

   roslaunch rgb_rgb_system_calibration full_evaluation.launch test_json:=$ATOM_DATASETS/rgb_rgb_system/test/dataset.json train_json:=$ATOM_DATASETS/rgb_rgb_system/train/atom_calibration.json

```terminal
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/ground_truth_frame_evaluation", line 306, in <module>
    mixed_dataset = getMixedDataset(train_dataset, test_dataset)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 791, in getMixedDataset
    for _, additional_tf in mixed_dataset['calibration_config']['additional_tfs'].items():
AttributeError: 'NoneType' object has no attribute 'items'
After filtering, will use 3 collections: ['000', '001', '002']
After filtering, will use 3 collections: ['000', '001', '002']
After filtering, will use 3 collections: ['000', '001', '002']
After filtering, will use 3 collections: ['000', '001', '002']
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/inter_collection_rgb_to_rgb_evaluation", line 184, in <module>
    mixed_dataset = getMixedDataset(train_dataset, test_dataset) 
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 791, in getMixedDataset
    for _, additional_tf in mixed_dataset['calibration_config']['additional_tfs'].items():
AttributeError: 'NoneType' object has no attribute 'items'
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/inter_collection_rgb_to_rgb_evaluation", line 184, in <module>
    mixed_dataset = getMixedDataset(train_dataset, test_dataset) 
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 791, in getMixedDataset
    for _, additional_tf in mixed_dataset['calibration_config']['additional_tfs'].items():
AttributeError: 'NoneType' object has no attribute 'items'
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/inter_collection_rgb_to_rgb_evaluation", line 184, in <module>
    mixed_dataset = getMixedDataset(train_dataset, test_dataset) 
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 791, in getMixedDataset
    for _, additional_tf in mixed_dataset['calibration_config']['additional_tfs'].items():
AttributeError: 'NoneType' object has no attribute 'items'
Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/rgb_to_rgb_evaluation", line 183, in <module>
    mixed_dataset = getMixedDataset(train_dataset, test_dataset) 
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 791, in getMixedDataset
    for _, additional_tf in mixed_dataset['calibration_config']['additional_tfs'].items():
AttributeError: 'NoneType' object has no attribute 'items'
[ground_truth_frame_evaluation-5] process has died [pid 48759, exit code 1, cmd /home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/ground_truth_frame_evaluation -train_json /home/mike/datasets/rgb_rgb_system/train/atom_calibration.json -test_json /home/mike/datasets/rgb_rgb_system/test/dataset.json -csf lambda x: True -sfr __name:=ground_truth_frame_evaluation __log:=/home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/ground_truth_frame_evaluation-5.log].
log file: /home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/ground_truth_frame_evaluation-5*.log
[rgb_left_to_rgb_right_charuco_pattern_evaluation-1] process has died [pid 48755, exit code 1, cmd /home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/rgb_to_rgb_evaluation -ss rgb_left -st rgb_right -train_json /home/mike/datasets/rgb_rgb_system/train/atom_calibration.json -test_json /home/mike/datasets/rgb_rgb_system/test/dataset.json -csf lambda x: True -sfr __name:=rgb_left_to_rgb_right_charuco_pattern_evaluation __log:=/home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/rgb_left_to_rgb_right_charuco_pattern_evaluation-1.log].
log file: /home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/rgb_left_to_rgb_right_charuco_pattern_evaluation-1*.log
[inter_collection_rgb_left_to_rgb_right_charuco_pattern_evaluation-2] process has died [pid 48756, exit code 1, cmd /home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/inter_collection_rgb_to_rgb_evaluation -ss rgb_left -st rgb_right -wf world -train_json /home/mike/datasets/rgb_rgb_system/train/atom_calibration.json -test_json /home/mike/datasets/rgb_rgb_system/test/dataset.json -csf lambda x: True -sfr __name:=inter_collection_rgb_left_to_rgb_right_charuco_pattern_evaluation __log:=/home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/inter_collection_rgb_left_to_rgb_right_charuco_pattern_evaluation-2.log].
log file: /home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/inter_collection_rgb_left_to_rgb_right_charuco_pattern_evaluation-2*.log
[inter_collection_rgb_left_to_rgb_left_charuco_pattern_evaluation-3] process has died [pid 48757, exit code 1, cmd /home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/inter_collection_rgb_to_rgb_evaluation -ss rgb_left -st rgb_left -wf world -train_json /home/mike/datasets/rgb_rgb_system/train/atom_calibration.json -test_json /home/mike/datasets/rgb_rgb_system/test/dataset.json -csf lambda x: True -sfr __name:=inter_collection_rgb_left_to_rgb_left_charuco_pattern_evaluation __log:=/home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/inter_collection_rgb_left_to_rgb_left_charuco_pattern_evaluation-3.log].
log file: /home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/inter_collection_rgb_left_to_rgb_left_charuco_pattern_evaluation-3*.log
[inter_collection_rgb_right_to_rgb_right_charuco_pattern_evaluation-4] process has died [pid 48758, exit code 1, cmd /home/mike/workspaces/catkin_ws/src/calibration/atom/atom_evaluation/scripts/inter_collection_rgb_to_rgb_evaluation -ss rgb_right -st rgb_right -wf world -train_json /home/mike/datasets/rgb_rgb_system/train/atom_calibration.json -test_json /home/mike/datasets/rgb_rgb_system/test/dataset.json -csf lambda x: True -sfr __name:=inter_collection_rgb_right_to_rgb_right_charuco_pattern_evaluation __log:=/home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/inter_collection_rgb_right_to_rgb_right_charuco_pattern_evaluation-4.log].
log file: /home/mike/.ros/log/7cf6f87a-9106-11ee-aa70-55c41639ea60/inter_collection_rgb_right_to_rgb_right_charuco_pattern_evaluation-4*.log

```

You can follow the readme at https://github.com/lardemua/atom/tree/noetic-devel/atom_examples/rgb_rgb_system to replicate

miguelriemoliveira, 2023-12-02:
Also crashes in rgb_rgb_system_chessboard 

    roslaunch rgb_rgb_system_chessboard_calibration full_evaluation.launch test_json:=$ATOM_DATASETS/rgb_rgb_system_chessboard/test/dataset.json train_json:=$ATOM_DATASETS/rgb_rgb_system_chessboard/train/atom_calibration.json

manuelgitgomes, 2023-12-04:
Unfortunately, this does not seem to be related with full_evaluation, but with the specific evaluation scripts.
I will work on it.

manuelgitgomes, 2023-12-04:
After a lot of work, I fully reformated evaluation to accept multiple patterns.

I have not tested various combinations as I cannot calibrate softbot2, which has almost all combination of sensors (#767)
When this is solved, I will test everything.

miguelriemoliveira, 2023-12-04:
Thanks @manuelgitgomes . Great work I will try it out in the atom examples and let you know of any bugs.

miguelriemoliveira, 2023-12-05:
I think we should close this one, and create new ones if / when bugs appear.

commit e712d5d8 (2024-04-29): Experiment Plan, lidar evaluation fix #750 #926

on the function modified, there were some keys switched arround, plus it
was lacking an argument.
closed 2023-12-05 by miguelriemoliveira