lardemua/atom #755: Create an atom_examples readme and a script to download all datasets at once.

miguelriemoliveira, 2023-12-02:


closed 2023-12-03 by miguelriemoliveira

manuelgitgomes, 2023-12-04:
Hello @miguelriemoliveira

Solved a bug where the program would because, for each system, a parent folder is needed.
Because I did not have them, it crashed.
Now the program creates the folder if needed.

Another problem, I download the rgb_rgb_system dataset first of the system's read me, and the dataset was not updated!

miguelriemoliveira, 2023-12-04:
> Solved a bug where the program would because, for each system, a parent folder is needed. Because I did not have them, it crashed. Now the program creates the folder if needed.

Thanks!

> Another problem, I download the rgb_rgb_system dataset first of the system's read me, and the dataset was not updated!

Did you try using the overwrite flag?

manuelgitgomes, 2023-12-04:
> Did you try using the overwrite flag?

Using the script, I had no problem, but I tried using the links of the readme and those were not correct.

commit 6d828265 (2023-12-03): Close #755
commit a0f601f7 (2023-12-03): Fix #755 extraction was creating an undesired subfolder