lardemua/atom #760: How to name atom examples

miguelriemoliveira, 2023-12-03:
rgb_hand_eye_system should be changed.

The logic for naming atom examples has been: 

\<modality\>_\<modality\>_\<configuration\>_system_\<pattern\>

which is a bit sluggish. Better to use :

\<modality\>_\<modality\>_\<configuration\>_\<pattern\>_system

or even:

\<modality\>_\<modality\>_\<additional distinctive characteristic\>_system

That means renaming the following atom examples:

rgb_hand_eye_system -\> rgb_in_hand_system

rgb_rgb_system_chessboard -\> rgb_rgb_chessboard_system

... -\> ...

but that results in very large names. So my decision is to go with smaller names, in the spirit of softbot and mmtbot. In this case, we define a name for the system, i.e.

Multi-modal test robot

and extract the acronym mmtbot. This is perhaps the best way, as it is the one we used successfully in the past.
We define the full name using the rules above, but then extract the acronym in mmtbot fashion.

Under this the examples would become:

Old name | Full new name | acronym
:---:|:---:|:---:
rgb_rgb_system | rgb_rgb_test_robot | rrtbot
rgb_depth_system |rgb_depth_test_robot | rdtbot
rgb_rgb_system_chessboard | rrctbot
rgb_hand_eye_system | rgb_in_hand_test_robot | rihtbot
eye_in_world_system | rgb_in_world_test_robot | riwtbot

@manuelgitgomes, let me know what you think. This should be well defined as we plan to develop more examples in the future. I will probably advance with this idea if you cannot comment in the next hour, as I will not have time to work on this during the week.

miguelriemoliveira, 2023-12-03:
Actually, I think its better to drop the **test_robot** and keep only the **robot**, i.e.:

Old name | Full new name | acronym
:---:|:---:|:---:
rgb_rgb_system | rgb_rgb_robot | rrbot
rgb_depth_system |rgb_depth_robot | rdbot
rgb_rgb_system_chessboard | rgb_rgb_chessboard_robot | rrcbot
rgb_hand_eye_system | rgb_in_hand_robot | rihbot
eye_in_world_system | rgb_in_world_robot | riwbot

miguelriemoliveira, 2023-12-03:
Thanks @Kazadhum .

manuelgitgomes, 2023-12-04:
Agree with @Kazadhum.

This table (or something similar) should be on the atom examples readme!

miguelriemoliveira, 2023-12-05:
Thanks to all. I will move forward with the proposed renaming.

closed 2023-12-06 by miguelriemoliveira