lardemua/atom #764: How will evaluation run when multiple patterns are present?

manuelgitgomes, 2023-12-04:
Hello @miguelriemoliveira.

I want to know your opinion on how to run evaluation for multiple patterns.
Will you continue to run for a pair of sensors and we would retrieve the values for all patterns?
Or should it be for only one pattern with a flag that chooses the pattern to be used?

This is because I am changing things in evaluation files, might as well add this...

miguelriemoliveira, 2023-12-23:
Hi @manuelgitgomes ,

I have completed #781 . Now we have a running example of a multi-pattern calibration. 

When you have time you can work on this. Just use the rihmpbot example readme.

miguelriemoliveira, 2024-01-14:
This is an old one, despite being opened on December 4th. It shows how much we've advanced in a month. Closing.

closed 2024-01-14 by miguelriemoliveira